DYefremov/DemonEditor #4: ModuleNotFoundError: No module named 'gi'

nospam999, 2020-01-11:
Hello,
Editor does not start and I get this error in Linux Mint 19.1 Tessa (Cinnamon). Can someone help?

`linus@linux ~/DemonEditor-master $ ./start.py 

Traceback (most recent call last):
  File "./start.py", line 2, in <module>
    from app.ui.main_app_window import start_app
  File "/home/linus/DemonEditor-master/app/ui/main_app_window.py", line 8, in <module>
    from gi.repository import GLib, Gio
ModuleNotFoundError: No module named 'gi'
`

DYefremov, 2020-01-11:
Hello.
I tested the launch of the program with Linux mint 19.1-cinnamon 64bit live iso. It works.
For first try to fix it by reinstalling PyGObject:
`sudo apt install --reinstall python3-gi`
if not helps, then look [here](https://github.com/mivoligo/Zeegaree/issues/6#issuecomment-343420559) for example.

nospam999, 2020-01-11:
Python was installed on this linux machine with Anaconda and I could not get installed python3-gi so far. Now I removed Anaconda (Python) and made 

`sudo apt install --reinstall python3-gi`

as you suggested and now I can start DemonEditor. Thank you very much for fast help !!!

DYefremov, 2020-01-11:
Ok. Happy to help.

closed 2020-01-11 by DYefremov