DYefremov/DemonEditor #12: Error Append services error: list index out of range

MarkusH12, 2020-03-08:
an idea why the error occurs when connecting to the box

DYefremov, 2020-03-08:
Hello.
Unfortunately, too little information. Have you edited the settings in other programs or through the web interface? To start with, show the output in the download dialog. Then it is desirable to get the following files from the '**data/enigma2/profile name**' folder: lamedb(lamedb5 if exist) and bouquets.tv, bouquets.radio. Or (and) best the last zip archive from the **'data/enigma2/backup/profile name'** name folder! For exact paths look in settings.
At the moment I can only assume that the problem when reading lamedb or lamedb5 (has experimental support and has not been fully tested)!

DYefremov, 2020-03-08:
Hello.
Unfortunately, too little information. Have you edited the settings in other programs or through the web interface? To start with, show the output in the download dialog. Then it is desirable to get the following files from the **'data/enigma2/profile name'** folder: lamedb(lamedb5 if exist) and bouquets.tv, bouquets.radio. Or (and) best the last zip archive from the **'data/enigma2/backup/profile name'**  folder! For exact paths look in settings.
At the moment I can only assume that the problem when reading lamedb or lamedb5 (has experimental support and has not been fully tested)!

tommiedepommie, 2020-05-02:
Hello,
After using the editor about three times i got the same error. I can't 'load' my saved channellist, can't 'ftp' the channellist into the program.  "list index out of range"
Pity, it sure has potential.

greetings tom

DYefremov, 2020-05-02:
Hi,
Which version are you using?
Try to run the program from the terminal as described above and show the output. Better yet, simply archive the current folder of that profile  where the error occurs(or better the entire data folder) and send it to me.
Best

tommiedepommie, 2020-05-02:
Hello, first of all, thank you for your quick response.
Second, I'm sorry.
My problem is solved. I had an IP address conflict in my network and one of my receivers acted strange. The error is gone now. Have a nice weekend.

grtzz tom

mischugo, 2020-05-07:
Same problem here. The network works fine.
FTP-Transfer -> Downloading Bouquets (or anything else) -> downloading -> Done. and PopUp
list index out of range.
Clicking ok. The window closed -> No list

DYefremov, 2020-05-07:
Hi.
First of all, install the latest version from [here](https://github.com/DYefremov/DemonEditor/raw/experimental-mac/dist/DemonEditor.app.zip) to the Application folder.
Then we delete the previous program settings. We give the command in the terminal:
`rm -R  .config/demon-editor/`
Run app from also from the terminal:
```
cd /Applications/DemonEditor.app/Contents/MacOS/
./DemonEditor
```
If app starts, then set up your profile for the receiver and try to download. If an error occurs, then you need to lay out the output from the terminal, as well as those files that you downloaded from the receiver, as described above. In addition, you can run the program with logging enabled (with the _-l_ key, described above). The _demon-editor.log_ should be created in the root of the folder.

DYefremov, 2020-05-07:
Hi.
First of all, install the latest version from here to the Application folder.
Then we delete the previous program settings. We give the command in the terminal:
rm -R .config/demon-editor/
Run app also from the terminal:

cd /Applications/DemonEditor.app/Contents/MacOS/
./DemonEditor

If app starts, then set up your profile for the receiver and try to download. If an error occurs, then you need to lay out the output from the terminal, as well as those files that you downloaded from the receiver, as described above. In addition, you can run the program with logging enabled (with the -l key, described above). The demon-editor.log should be created in the root of the folder.

DYefremov, 2020-05-07:
Hi.
First of all, install the latest version from [here](https://github.com/DYefremov/DemonEditor/raw/experimental-mac/dist/DemonEditor.app.zip) to the Application folder.
Then we delete the previous program settings. We give the command in the terminal:
`rm -R .config/demon-editor/`
Run app also from the terminal:
```
cd /Applications/DemonEditor.app/Contents/MacOS/
./DemonEditor
```
If app starts, then set up your profile for the receiver and try to download. If an error occurs, then you need to lay out the output from the terminal, as well as those files that you downloaded from the receiver, as described above. In addition, you can run the program with logging enabled (with the -l key, described above). The demon-editor.log should be created in the root of the folder.

DYefremov, 2020-05-07:
Hi.
First of all, install the latest version from [here](https://github.com/DYefremov/DemonEditor/raw/experimental-mac/dist/DemonEditor.app.zip) to the Application folder.
Then we delete the previous program settings. We give the command in the terminal:
`rm -R .config/demon-editor/`
Run app also from the terminal:
```
cd /Applications/DemonEditor.app/Contents/MacOS/
./DemonEditor
```
If app starts, then set up your profile for the receiver and try to download. If an error occurs, then you need to lay out the output from the terminal, as well as those files that you downloaded from the receiver, as described above. In addition, you can run the program with logging enabled (with the _-l_ key, described above). The _demon-editor.log_ should be created in the root of the folder.

mischugo, 2020-05-07:
Thank you so much for your support.

It's totally crazy. I unzipped the zip file with another program. And now I have an app that opens immediately with a double click.

Unbelievable

1.

2.

Should I make preparations (Terminal) now or can I start the app right away?
Is there a update feature for upcomming versions or must i install every time a new version

I am not familiar with these editors and all the features of a box. I haven't dealt with that so far. So, i can now tryout what it can do for me.

Best and thanks again

Maik

mischugo, 2020-05-07:
It’s me again

But the issue ist still there

Cheers,

Maik

DYefremov, 2020-05-07:
Hi
Since the program is in the process of active development, there is no possibility of automatic updating yet. In addition, if problems arise, you must have a certain level of knowledge, at least know what a terminal is. If this is all ok, then I can try to help you in this discussion.
If not, then I will delete this discussion later, and you can write to me on the email indicated in my profile. And in the presence of free time I  can try to help you.

DYefremov, 2020-05-07:
Hi
Since the program is in the process of active development, there is no possibility (for macOS) of automatic updating yet. In addition, if problems arise, you must have a certain level of knowledge, at least know what a terminal is. If this is all ok, then I can try to help you in this discussion.
If not, then I will delete this discussion later, and you can write to me on the email indicated in my profile. And in the presence of free time I can try to help you.

mischugo, 2020-05-07:
Thanks a lot. Sounds good.

Best

Maik

DYefremov, 2020-05-08:
Previous conversation is cleared, due to weak intersection with discussion.

If such an error occurs, please do the following (for [macOS](https://github.com/DYefremov/DemonEditor/tree/experimental-mac) version):
Install the latest version from [here](https://github.com/DYefremov/DemonEditor/raw/experimental-mac/dist/DemonEditor.app.zip)  to the Application folder . **Allow the launch of the program**, if necessary, in the security settings!

Then remove the previous program settings. Give the command in the terminal:

`rm -R .config/demon-editor/`

Run app also from the terminal (macOS):
```
cd /Applications/DemonEditor.app/Contents/MacOS/
./DemonEditor
```
For  Linux: 

  `./start.py`

in the root of app folder, or 

` demon-editor`

if the program is installed from a _*.deb_ package or _PPA_ repository.

After app starts, then set up your profile for the receiver and try to download. If an error occurs, then lay out the output from the terminal, as well as those files that you downloaded from the receiver, as described above. In addition, you can run the program with logging enabled (with the _-l_ option, described above). The _demon-editor.log_ should be created in the root of the folder.

The following information is also desirable:
1. The build date of the program (maсOS) or version (brаnch).
2. The used enigma2 image.
3. Have the settings been edited in other programs.

DYefremov, 2020-05-08:
Previous conversation is cleared, due to weak intersection with discussion.

DYefremov, 2020-05-08:
Previous conversation is cleared, due to weak intersection with discussion.

Perhaps [this](https://github.com/DYefremov/DemonEditor/commit/2e34568f743e46f3651272604eea640f712620dc) fix will help to prevent this error.

mischugo, 2020-05-08:
Thanks again.

Don't stress, I'm in no hurry with this whole thing.

It's not easy. With the terminal (new shell) in Catalina, it doesn't fit everything. Especially because I can't see in which directory I have to be right now to enter the commands
Anyway.

I have deleted corresponding data / folders in my user directory.
Download from the receiver works.

But then it becomes difficult. Closing the window not possible. No idea why. Then i start screenflow to capture (screencast) whats going on.
Suddenly, the Demon Editor window is ready to close … Crazy

Using:

1. MacOS Catalina Version 10.15.4 (19E266)
2. Enigma 2, Octagon SF4008, Image OpenATV 6.4
3. I have no editor because nothing works with Catalina

Cheers

Maik

DYefremov, 2020-05-08:
> But then it becomes difficult. Closing the window not possible. No idea why. Then i start screenflow to capture (screencast) whats going on. Suddenly, the Demon Editor window is ready to close … Crazy
Hi, Maik
Please create, if you can, for this case, a separate issue. I will close it!
Thanks!

DYefremov, 2020-05-08:
> But then it becomes difficult. Closing the window not possible. No idea why. Then i start screenflow to capture (screencast) whats going on. Suddenly, the Demon Editor window is ready to close …

Hi, Maik
Please create, if you can, for this case, a separate issue. I will close it!
Thanks!

DYefremov, 2020-05-09:
Hi,
I returned the [older](https://github.com/DYefremov/DemonEditor/raw/experimental-mac/dist/DemonEditor.app.zip) build. Try with it.
Best

DYefremov, 2020-05-13:
> But then it becomes difficult. Closing the window not possible.

Hi, Maik
Please try with the new [build](https://github.com/DYefremov/DemonEditor/raw/experimental-mac/dist/DemonEditor.app.zip). If it does not help, create a separate issue for this case.
Thanks!

closed 2020-05-13 by DYefremov