DYefremov/DemonEditor #16: import sat&channels from websites

data84, 2020-03-20:
hi
This is a great app for linux :)

can you, in the future, add the option to import satellites and channels from lyngsat.com, kingofsat.net or others websites?

It would be a useful option for creating new lists, but it is not available in many applications of this type.
 Probably only the paid version of "echannelizer" has this option working now.

/sorry for my English :) 
 Regards data84

DYefremov, 2020-03-21:
Hi.
Look carefully in the satellites edit tool. This feature already exists!
Best regards

[image]

DYefremov, 2020-03-21:
Hi.
Look carefully in the satellites edit tool. This feature already exists!

[image]

I’ve already thought about importing channels, but I have postponed it for now. Perhaps in future versions I’ll add it.
My English is also far from perfect! :)
Best regards

data84, 2020-03-21:
Yes yes , 
I see the sat updates from lyngsat or flysat. 
I tested it. It works very good , but it can be useful to download channels as well from web: 
1. to one or several transponders
2. to one several satellites 

.....:)
And very very very later;) version,  maybe you add the function conversion to other channel list formats

DYefremov, 2020-03-21:
>maybe you add the function conversion to other channel list formats

Hi
If you mean the conversion from ver .4 to lamedb5, then this functionality is already there.
Just load the list of channels in the old format, and then turn on ver. 5 support in the settings and save! Later I will add the menu item "Save as ...".  Usually, users rarely use this feature (just uses one default format), including me. :)

Best regards

data84, 2020-03-21:
I mean conversion to a different format than e2, but it is not important now :)

DYefremov, 2020-03-21:
Ok. Then have a good tests! :+1:

DYefremov, 2020-11-02:
>it can be useful to download channels as well from web:
>     1. to one or several transponders
>     2. to one several satellites

Added **experimental** implementation for this  feature. So far, only basic functionality. Will be gradually improved.
[image]

closed 2020-11-24 by DYefremov