DYefremov/DemonEditor #21: MacOS: ValueError: not enough values to unpack (expected 2, got 1)

nano1969, 2020-05-03:
Hi,
I get the following error messages when I want to load the data from the receiver:
Reading data error: Traceback (most recent call last):
  File "app/ui/main_app_window.py", line 1102, in update_data
  File "app/eparser/__init__.py", line 30, in get_bouquets
  File "app/eparser/enigma/bouquets.py", line 14, in get_bouquets
  File "app/eparser/enigma/bouquets.py", line 124, in parse_bouquets
  File "app/eparser/enigma/bouquets.py", line 96, in get_bouquet
ValueError: not enough values to unpack (expected 2, got 1)

Any ideas?

DYefremov, 2020-05-03:
It looks as if you have channels with an additional name (pseudonym) specially for this bouquet. This is quite normal practice, this program also allows you to do this. Have you edited your bouquets in other programs?  It also highly advisable to get your bouquet files to determine the source of the problem.

DYefremov, 2020-05-03:
Hi,
It looks as if you have channels with an additional name (pseudonym) specially for this bouquet. This is quite normal practice, this program also allows you to do this. Have you edited your bouquets in other programs?  It also highly advisable to get your bouquet files to determine the source of the problem.

nano1969, 2020-05-03:
Hi DYefremov,

I used iDreamX before.
I loaded a new channel list and now it works! Thanks

closed 2020-05-03 by nano1969