DYefremov/DemonEditor #27: Jump markers when counting channel numbers

fneinewtec, 2020-06-17:
Hello Dmitriy,
I concur DemonEditor is for us Ubuntu users the greatest channel editor!
Would it be possible to jump the markers when counting channel numbers?
On my openpli receiver, the channel numbers do never concur with the ones in the editor, because an extra number is added for every marker.
Kind regards,
Frank

DYefremov, 2020-06-17:
Hi Frank,
Yes, it's possible. I plan to add the filtering and sorting feature to the bouquet list and I will try to take this suggestion into account.
Best,
Dmitriy

closed 2020-06-22 by DYefremov

DYefremov, 2020-06-22:
Hi Frank,
Yes, it's possible. [Done!](https://github.com/DYefremov/DemonEditor/commit/1cec96d2b5c1a829282dfb0e355c48eddbb25acb)

Best,
Dmitriy