DYefremov/DemonEditor #34: Feature: show SID as decimal instead of hex

vpont, 2020-09-08:
In services' list it would really help me when comparing services with KingOfSat if one could choose between showing services' SIDs as decimal or hex numbers.

DYefremov, 2020-09-08:
I think it will be useful. As an option, I can offer a context (pop-up) menu for the column header with the choice of the desired display.

DYefremov, 2020-09-11:
Hi. Hey. I made a test implementation of this function.Perhaps she will suit you.  But...
[image]
[DemonEditor_sid_as_decimal.zip](https://github.com/DYefremov/DemonEditor/files/5211444/DemonEditor_sid_as_decimal.zip)

Unfortunately on big data (> 5000 rows in the main list) it can be slow. I tried several options and came to the conclusion that it would be optimal to do the following:
**Option 1.**
We just add the options to show in tooltips as decimal!
**Option 2.**
We convert the values when loading data and again show them in tooltips. But here it is already desirable to restart the program!
**Option 3.**
Or we leave this test option(with popup menu) that is in the attachment.

Personally, I tend to the **first** option. I will be glad for your thoughts.

vpont, 2020-09-12:
Hi.
What about keeping it simple and just add the SID in the tooltip in both formats, like this:
[image]
No extra config options or reloads.

DYefremov, 2020-09-12:
Hi. Yes, that's a **good option**! We only need to select a display location in the main list. Alternatively after the service type or on the next line. I think the first option.

vpont, 2020-09-12:
Maybe after the service type, in order to maintain consistency with the main window list?

DYefremov, 2020-09-12:
Yes, it look ok too. Please [test it](https://github.com/DYefremov/DemonEditor/archive/development.zip).

vpont, 2020-09-12:
Looks good to me!
Just that extra space in front of the decimal value:
[image]

vpont, 2020-09-12:
And the hex value would look cleaner in upercase

DYefremov, 2020-09-12:
Ok. [Done](https://github.com/DYefremov/DemonEditor/archive/development.zip)!

vpont, 2020-09-13:
Looks good, thank you!!

closed 2020-09-13 by vpont