DYefremov/DemonEditor #39: Update 0.4.8 Pre-alpha (Build: 20200507)

mischugo, 2020-11-13:
Sorry for the question, but I am not a developer... And I have no idea how else to contact you. Is there a way to update an earlier version? If so, how do I do that

Thank you so much for this great piece of software

Best - Maik

DYefremov, 2020-11-13:
Hello, Mike.

After version 1.0.0 you can download the ready-made packages as a _*.dmg_ file from the [releases](https://github.com/DYefremov/DemonEditor/releases) page.
[image]

Here's the[ latest version](https://github.com/DYefremov/DemonEditor/releases/download/1.0.1-b1-macOS/DemonEditor.dmg) for macOS. It is advisable to uninstall the previous version before installing!
For more info, you can read [here](https://github.com/DYefremov/DemonEditor/tree/experimental-mac).

Best
Dmitriy

mischugo, 2020-11-13:
Hello Dimitriy,

thank you so much for teh fast response.
I was on that page, but it looks other then your screenshot

There was no *.dmg file

That’s why i asked in github

Is it like this, that i only delete the older version or is there a must for any uninstaller?

Best

Maik

DYefremov, 2020-11-13:
> Is it like this, that i only delete the older version or is there a must for any uninstaller?
Just move to trash from the Applications folder.

Best
Dmitriy

mischugo, 2020-11-13:
Thanks a lot!!!!!

closed 2020-11-14 by DYefremov