DYefremov/DemonEditor #41: too many values to unpack

rakanbakir, 2021-01-04:
version: 1.0.3 Beta (Build: 20201225)
OS: macOS 10.15.7

When ever I try to import data from my receiver I got the below error

`$ ./DemonEditor -d on
Debug mode is on.
Reading data error: Traceback (most recent call last):
  File "app/ui/main_app_window.py", line 1513, in update_data
  File "app/eparser/__init__.py", line 15, in get_services
  File "app/eparser/enigma/lamedb.py", line 16, in get_services
  File "app/eparser/enigma/lamedb.py", line 85, in parse
  File "app/eparser/enigma/lamedb.py", line 165, in parse_v5
ValueError: too many values to unpack (expected 2)`

any idea what is happening?

DYefremov, 2021-01-04:
Hi,
First, turn off support for lamedb **version 5** in the settings! It looks like a parsing error for this format. And try to load the settings again. If possible, send me on email from my profile 2 files: **lamedb**, and especially **lamedb5**!

Best

rakanbakir, 2021-01-04:
I have disabled lamedb5, and it worked, thx.

closed 2021-01-04 by rakanbakir