DYefremov/DemonEditor #42: Programs with list of alternatives do not appear

Schorsch99, 2021-01-04:
version: 1.0.3 Beta
Linux Mint 20 Ulyana

In my Favourites bouquet I have mixed "normal" channels and several channels with a list of alternatives. These channels with alternatives do not appear in DemonEditor and additionally the numbering for these channels is not considered, so that the numbers of the "normal" channels are incorrect. 
Here is my (shortened) "userbouquet.favourites.tv":

#NAME Favoriten
#SERVICE 1:64:1:0:0:0:0:0:0:0::Hauptprogramme
#DESCRIPTION Hauptprogramme
#SERVICE 1:134:1:0:0:0:0:0:0:0:FROM BOUQUET "alternatives.das_erste_hd.tv" ORDER BY bouquet
#SERVICE 1:134:1:0:0:0:0:0:0:0:FROM BOUQUET "alternatives.zdf_hd.tv" ORDER BY bouquet
#SERVICE 1:64:2:0:0:0:0:0:0:0::Dritte
#DESCRIPTION Dritte
#SERVICE 1:134:1:0:0:0:0:0:0:0:FROM BOUQUET "alternatives.swr_bw_hd.tv" ORDER BY bouquet
#SERVICE 1:134:1:0:0:0:0:0:0:0:FROM BOUQUET "alternatives.rbb_berlin_hd.tv" ORDER BY bouquet
#SERVICE 1:64:3:0:0:0:0:0:0:0::Spartenprogramme
#DESCRIPTION Spartenprogramme
#SERVICE 1:134:1:0:0:0:0:0:0:0:FROM BOUQUET "alternatives.kika_hd.tv" ORDER BY bouquet
#SERVICE 1:64:4:0:0:0:0:0:0:0::Privatsender
#DESCRIPTION Privatsender
#SERVICE 1:0:1:445C:453:1:C00000:0:0:0:
#SERVICE 1:0:1:4460:453:1:C00000:0:0:0:

Here is one of the alternative channel lists ("alternatives.das_erste_hd.tv")
#NAME Das Erste HD
#SERVICE 1:0:19:283D:3FB:1:C00000:0:0:0:
#SERVICE 1:0:1:6DCA:44D:1:C00000:0:0:0:
#SERVICE 1:0:1F:301:3003:2114:EEEE0000:0:0:0:

In DemonEditor it appears like this:

Num Service                     Model           Pos.
         Hauptprogramme     MARKER
         Dritte                        MARKER
         Spartenprogramme MARKER
         Privatsender            MARKER
1       SAT.1                       TV               19.2
2       SAT.1 Gold              TV                19.2
etc.

And this is what it should be:

Num Service                 Model          Pos.
    Hauptprogramme      MARKER
1   Das Erste HD           TV              19.2
2   ZDF                          TV              19.2
    Dritte                         MARKER
3   SWR BW HD           TV               19.2
4   RBB Berlin HD        TV                19.2
    Spartenprogramme  MARKER
5   KIKA HD                  TV               19.2
    Privatsender        MARKER
6   SAT.1                       TV              19.2
7   SAT.1 Gold               TV             19.2
etc.

Did I miss a setting or is handling of alternative channel lists not implemented(yet)?

DYefremov, 2021-01-04:
Hi,
> handling of alternative channel lists not implemented(yet)?

Yes you are right! At the moment, only alternative names of services in the bouquet lists have been implemented!
You are the first who asked about such an feature. I'm not even sure if this option works on my image. :)
But if you really need this feature, then I can implement it. I just need a complete set of your working settings and a desire to test!

Best

Schorsch99, 2021-01-04:
Hmmm. You are right, maybe this feature is actually not needed anymore. I had been using this method on my older receiver, which only had 2 tuners. Now with unicable and FBC (8 virtual tuners), I do not really need this any more and I will remove it :-)

But since Dreamboxedit has this feature, you may want to implement it anyway. So if I can help, let me know what data you need. I am also willing to do some testing :-)

DYefremov, 2021-01-04:
I also have 2x DVB-S2 + DVB-T2/C. :) 
As for the function, I'll try to implement it all the same. Please send me your full set of the settings(with lamedb and all bouquet files) for a sample on email from my profile. Thank you!

Best

closed 2021-02-14 by DYefremov