DYefremov/DemonEditor #46: Receiving Picons quits after getting links

tschmidt01, 2021-04-02:
Version: 1.0.6 Beta, Mac OS Build

After selecting a satellite and loading the providers, the "Receive picons" action quits after getting the links and thus does not download any images. I've tried it for different packages, but for none of them the picons get downloaded.

[image]

DYefremov, 2021-04-03:
Hello,
Please try [this](https://github.com/DYefremov/DemonEditor/releases/download/1.0.7-b1-macOS/DemonEditor.dmg) build. 
[image]

tschmidt01, 2021-04-03:
Works, thanks a lot!

DYefremov, 2021-04-03:
Hello,
I added another small fix and updated the build. Let me know if something is wrong.
[image]

tschmidt01, 2021-04-03:
There might be another problem with downloading the picons, when you select all providers the same thing happens as what I've opened the issue for (except is does not quit but gets stuck in "please wait"):

[image]

DYefremov, 2021-04-03:
Yes you are right. There was an error loading picons for some single channels. Please [test it](https://github.com/DYefremov/DemonEditor/releases/download/1.0.7-b1-macOS/DemonEditor.dmg). Hopefully it's fixed now. :) 
Since LyngSat can change the page format and content at any time, it is possible to repeat this case.

tschmidt01, 2021-04-03:
Works perfectly, will open a new issue if I find something :) Thanks a lot!

closed 2021-04-03 by tschmidt01

DYefremov, 2021-04-04:
Thanks. I'll be glad to any comments and suggestions.