DYefremov/DemonEditor #49: Downloaded picons should be offered as options in the attach picon menu command

dikonov, 2021-10-03:
В менеджере пиконов имеется функция загрузки значков из Интернета. Полученные таким образом значки "пиконы" сохраняются на диск, и среди них есть те, которые следует связать с найденными поиском каналами. Они видны в менеджере пиконов в поле "Проводника" "Назначение". Однако, меню "Пикон/Привязать" сразу открывает диалог выбора файла на диске, где трудно найти нужный. Картинок могут быть тысячи, а предпросмотра выбор файлов не имеет. В результате, привязка пиконов к новым каналам оказывается трудоемким занятием.

Просьба:
1)  Добавить новую команду в меню "Пикон", открываемом правой кнопкой мыши в списке каналов. Новый пункт меню (активный только при наличии пиконов в поле "Назначение" менеджера пиконов) показывал бы панель доступных пиконов из менеджера. Уже существующую команду с диалогом выбора файла можно переименовать в "Привязать файл".

2) В новом меню визуального подбора пиконов автоматически выделять тот, чье имя файла совпадает с названием выбранного канала или его транслитерацией (НТВ = нтв.png/ntv.png/Нтв.png...) либо с условным именем в формате энигмы (1_0_19_FD2_AA_1_FFFF0000_0_0_0.png). Если выбрано несколько каналов, то так можно связывать сразу всю группу. 

Кроме того:
* "Проводник" в менеджере пиконов непонятно для чего имеет два поля "Источник" и "Назначение". Там нет команд копирования между этими местами, а набор команд в меню значков "Загрузить в ресивер/из ресивера" одинаков. "Назначение" явно показывает каталог на ПК, а не то, что есть на диске ресивера.
  
* При загрузке пиконов с picon.cz у меня возникает ошибка распаковки 7zip (7zr). У меня стоит p7zip, и такие архивы открываются в других программах.

DYefremov, 2021-10-03:
Приветствую.
Практически все, что вы описали, есть в наличии. Просто, к сожалению, возможно это не совсем очевидно.  В архиве старый скринкаст для  macOS версии.  НО, думаю должно стать понятнее. Ошибка в конце уже исправлена :)
[DemonEditor_Picons_Manager.mp4.zip](https://github.com/DYefremov/DemonEditor/files/7272952/DemonEditor_Picons_Manager.mp4.zip)

 С 7zip нужно разбираться. Укажите свой дистририбутив и версию программы. В дистрибутивах, основанных на Ubuntu, 7zr поставляется с пакетом p7zip.

luxorko, 2021-10-04:
where shoud be download v2?

DYefremov, 2021-10-04:
Select the development-2.0 branch in the left drop-down list, and then click on the green "Code" button on the right and select "Download ZIP". [Direct link!](https://github.com/DYefremov/DemonEditor/archive/refs/heads/development-2.0.zip) 
**Important.** This version can reset your program settings to default!

luxorko, 2021-10-05:
Thank you. I have beta v1.0.10-a-win; development-2.0. and copy over and re-writing
when test it found it "download service update" - its will create lamedb, but after copy to sat. nothing vissible.
tested with other program "E-chan......"and get error "invalid format"

DYefremov, 2021-10-05:
Hi, 
This is not the right way. You need to use [MSYS2](https://www.msys2.org/) to run from source or just look at [releases](https://github.com/DYefremov/DemonEditor/releases). Sometimes I post here ready-made test builds for Windows.  Beside this, I don't think it's worth discussing Windows builds here. I recommend that you take your time and wait for a ready-made solution for your OS. 

Best

dikonov, 2021-10-13:
В ALTLinux пакет p7zip содержит:

$ rpm -ql p7zip | grep -v doc
/usr/bin/7z
/usr/lib64/p7zip
/usr/lib64/p7zip/7z.so
/usr/lib64/p7zip/7zCon.sfx
/usr/lib64/p7zip/Codecs
/usr/lib64/p7zip/Codecs/Rar.so
/usr/share/man/man1/7z.1.xz

т.е. для распаковки 7zip есть только 7z

DYefremov, 2021-10-14:
Если речь про такое расположения подсказки, то вполне рабочий вариант.

[image]

[image]

Касаемо spec файла для  ALT. Думаю будет полезно.

dikonov, 2021-10-14:
[demon-editor-ALTLinux spec.zip](https://github.com/DYefremov/DemonEditor/files/7348771/demon-editor-ALTLinux.spec.zip)
Вот только пользоваться версией 2 не получается. Старые настройки от 1.0.10 она автоматически не открывает, а настроить заново нельзя. Диалог настройки недоступен.

DYefremov, 2021-10-14:
Спасибо. Планирую навести порядок в сборочных скриптах для разных платформ. Заодно добавлю и ваш скриптик. Не совсем понял зачем отдельный патч для стартового скрипта (можно было заглянуть в каталог [deb](https://github.com/DYefremov/DemonEditor/tree/development-2.0/deb)), но не соль. Думаю оптимально было бы использовать готовый *deb пакет котрый я выкладываю вместе с релизамами (вот [пример](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=demoneditor-bin) для Arch).  Но в любом случае, как пример, будет полезен.

DYefremov, 2021-12-03:
Реализовал п.1 и п.2, но с некоторыми отличиями. При выборе пункта "Привязать", из контекстного меню, будет открыта вкладка работы с пиконами. Можно, как и предлагается, выбрать сразу несколько сервисов, но, вместо выделения, будет ативирована **фильтрация** по всем выделленым сервисам! Дополнительным бонусом такой реализации является возможность фильтарции и поиска пиконов сразу для нескольких сервисов, использую символ **'|'** как разделитель (как по имени файла так и по имени сервиса). 

[image]

Если  будут замечания или дополнительные предложения по данной реализации буду рад обсудить.

closed 2021-12-04 by DYefremov