DYefremov/DemonEditor #50: Win64

EnoSat, 2021-10-19:
[image]
After download show error

DYefremov, 2021-10-19:
If possible, zip the ~"Demon Editor\data" folder and send it to me.

DYefremov, 2021-10-19:
Hi. If possible, zip the ~"Demon Editor\data" folder and send it to me.

DYefremov, 2021-10-19:
Sorry. Apparently not quite correctly put it. I need this folder.
[image]

DYefremov, 2021-10-19:
Ok. Thanks. The lamedb file (for Enigma2) contains an option flags with an hex value. This is usually a numeric value. 
[image]

I'm not sure if this value is valid. Hopefully someone will comment. For now, I will consider it an bug in the program.

EnoSat, 2021-10-20:
[image]
Hi, this build work OK ;)

DYefremov, 2021-10-20:
Fine! There is only this left: 
> After download in NHD ok (but not show terestrial.xml list)
It's not entirely clear what you mean.

DYefremov, 2021-10-21:
You were right. It was a small bug. And the old one.  Displaying of the DVB-T/T2 system has been implemented for a long time, I just forgot. :)

[image]