DYefremov/DemonEditor #54: import files

luxorko, 2021-12-16:
current version not support add bouquet.tv and m3u files from folder on pc.
IPTV channels list created on the userbouquet.tv format 
i thing some versions before had that future.could you add again please with m3u 
at the moment we can transfering only whole list where lamedb is present

DYefremov, 2021-12-16:
Hi,
It's not entirely clear what you mean. 
1. Single import of bouquets is available through the context menu of the bouquets list. 
2. Through the menu: File-> Import-> Bouquet.
3. Selective import of several bouquets through the menu :  File-> Import-> Bouquets ans services.

In addition, import is available via _drag and drop_. For bouquets, drag a single bouquet or folder onto the list of bouquets. For bouquet and services on the main list of services!  If you are using OS other than Linux, this function may not work correctly yet. Perhaps you mean this? 

> only whole list where lamedb is present

Just create a new configuration: File-> New empty configuration. Create a new bouquet and import _*.m3u_. Then, if necessary, you can export this bouquet through the context menu of the list of bouquets "Save As".  Or you meant something else?

Best

luxorko, 2021-12-16:
1 - Drag and drop  - not working - 
2 - "File-> Import-> Bouque"... folder with userbouquet.tv not opening 
3 -" File-> New empty configuration. Create a new bouquet and import *.m3u" - 
this option (Create a new bouquet and import *.m3u - not exist-cannot find ( where could be?)

only can opening folder where is userbouquet. tv with lamedb

DYefremov, 2021-12-16:
Apparently we still do not understand each other. :)

[image]
[image]
[image]
or:
[image]
[image]

> Drag and drop - not working -

> If you are using OS other than Linux, this function may not work correctly yet.

What operating system are you using (Windows, Linux, macOS)? 

I've tested imports on Linux and Windows 10. Everything works except drag and drop in Windows (BUT it is written about it above). 
Please describe in steps what you want to do. It might be easier to understand.

luxorko, 2021-12-16:
[image]
icon on the Demon are too small, my eyes are old.
could you make them bigger please?

find it bug:
1 - epty new configuration 
2 - import m3u files
3 - nothing happend (non choise for pick m3u files)
4 - first need load any bouquet and them you can load m3u files

rest working like charm. thank you pictures

luxorko, 2021-12-16:
during downloading picons, VPN must be deactivated othervise download not start
it is normal?

luxorko, 2021-12-19:
i think you could close it now. thank you

luxorko, 2021-12-19:
one more thing. 
you got downloading one setup for Hispasat 30W and second from Eutelsat 36 B,C 
both satellite are impossible to scan by one satellite dish
how to joint both lamedb together? its possible?
think will be usefull for watching IPTV and get EPG
hope you know what i thing

DYefremov, 2021-12-19:
> how to joint both lamedb together? its possible?

Yes, sure. Through the import function (File -> Import -> Bouquets and services). You can load one configuration then import the next one.  Or via drag and drop (In the current version may not work in Windows. ). For an example, see [here](https://www.youtube.com/watch?v=MybSevjQiN0).  I recommend watching the entire [playlist](https://www.youtube.com/playlist?list=PLB16nAGnuDczIx8NXyJYHuA9EPtDTRPHq). Many questions will probably disappear. 

> think will be usefull for watching IPTV and get EPG
Yes, it is planned, but I cannot say when it will be implemented yet.

luxorko, 2021-12-19:
think you can close this thread. thank you for instruction how to use this fantastic plugin

closed 2021-12-19 by DYefremov