DYefremov/DemonEditor #56: Просьба добавить сохранение m3u целого букета спутниковых каналов из редактора

dikonov, 2021-12-24:
Многие ресиверы с Энигмой поддерживают раздачу спутниковых каналов с тарелки в виде IPTV (работают как личный сервер IPTV). Для использования функции нужно создать свой список ссылок на каналы в формате m3u. В старых версиях OpenATV это можно было сделать в веб-интерфейсе OpenWebIF, но теперь это не работает. Можно получать ссылки только для одного канала за раз, а потом вручную их склеивать в общий список. 

Просьба добавить в редактор функцию сохранения всего букета в виде m3u для удаленного просмотра IPTV со своего ресивера.
Ее можно поместить в меню действий с букетом, рядом с "Save As" или оформить как формат сохранения.

Вместе с тем, в меню правой конки мыши на каналах есть пункт IPTV/Export ro m3u, который не работает.Выдается дурацкая ошибка "Текущий список не содержит потоков IPTV". И что из этого следует? Ресивер сам умеет эти потоки создавать. Если есть IPTV каналы перемешанные с DVB, то они уже были импортированы из списка. Полезность функции извлечения обратно уже известных импортированных ссылок IPTV явно ниже чем получение сетевой ссылки на DVB канал.

DYefremov, 2021-12-24:
> "Текущий список не содержит потоков IPTV".

Это было сделано специально для экспорта **только** IPTV списков. 

> Полезность функции извлечения обратно уже известных импортированных ссылок IPTV явно ниже чем получение сетевой ссылки на DVB канал.

Не поверите, но именно такое поведение также востребовано. Достаточно мало пользователей смешивают DVB и IPTV каналы. 

> Ее можно поместить в меню действий с букетом, рядом с "Save As"

Да, думаю можно сделать. Как вариант "Save As *m3u". Поэкспериментирую на досуге.

DYefremov, 2021-12-28:
Добавил запрашиваемый функционал. Надеюсь, что будет действительно полезно.

[image]
[image]

closed 2021-12-29 by DYefremov

dikonov, 2022-01-07:
Спасибо!
Сохранение в m3u позволяет смотреть свою тарелку там, где не работает редактор: на телефоне, планшете, телевизоре в другом городе и т.п. Так что это важная функция!