DYefremov/DemonEditor #57: App force closes on macOS Monterey 12.1

jffluis, 2022-01-03:
As stated the app force closes on macOS Monterey 12.1.

DYefremov, 2022-01-03:
Hello. Is [Homebrew](https://brew.sh/) installed on the system? There may be a library conflict. If so, try to temporarily remove all **GLib**-related packages. Alternatively, remove all packages completely, if possible.

 `brew remove --force $(brew list)`

jffluis, 2022-01-03:
I will try it now. TY.
If have some updates i reach to you

jffluis, 2022-01-03:
UPDATE: Updated brew and all fine now. Thank you.

DYefremov, 2022-01-03:
I also ran a small test on a virtual machine.
 The problem, as I wrote, is in conflict with the **GLib** part (libgio). If you are Homebrew user, then you can run the program from source as on Linux. If not, and you are using a package from _*.dmg_, then Homebrew and something else do not need to be installed. Everything is already included in the package! 

Best

closed 2022-01-04 by DYefremov