DYefremov/DemonEditor #58: Playback VLC, not found or installed

matthias-zivim, 2022-01-22:
Hello and thank you for your continued development!!!

I use DemonEditor 2.0.3 beta
I use VLC for macOS (Apple Silicon) M1

When I press play, No VLC is found. Check if installed

Can you help.

DYefremov, 2022-01-22:
Hello!
VLC player must be installed by default in ''Applications''. Since DemonEditor is built for Intel platform, it may be necessary to install the[ Intel version](https://get.videolan.org/vlc/3.0.16/macosx/vlc-3.0.16-intel64.dmg) of VLC from the official website. 

P.S. At the moment, there have been no reports of non-working  stream playback, including users of the M1 CPU.

luxorko, 2022-01-23:
regarding player:
when opening iptv channels and stream not work: 
Using only MPV player - if stream not work player stay black and get information in ( RED panel) on the bottom of program with message - "stream not avaible...."
issue - need to close player and this "information panel" - in only closing one; other stay ON and cannot opening - testing other stream
could you merge and when close one of tham will be closed both please.

DYefremov, 2022-01-23:
I think this is not entirely correct. Perhaps when the player is closed. It would probably be better to close the error message when selecting the next channel(stream) from the list. I should think. There is also a separate tool for testing and automatically removing unavailable streams. Select "Remove all unavailable" in the IPTV menu!

luxorko, 2022-01-23:
"Remove all unavailable" in the IPTV menu
didnt know above - just checked and is work.
it checking in all streams in all bouquets, not just in the opened bouquet.

closed 2022-01-27 by DYefremov