DYefremov/DemonEditor #59: channels ID list

luxorko, 2022-01-23:
previous version managet created " channels ID list" 
could you let me know how to make again please
- Regarding channels ID name "channel id="57"" - could pe possible keep original channels names from enigma-bouquet instead "number";please
- or have both option 

`<?xml version="1.0" encoding="utf-8"?>

<channels>
<channel id="57">4097:0:1:171:0:0:0:0:0:0</channel> 
<channel id="470">4097:0:1:1418:0:0:0:0:0:0</channel> 
<channel id="289">4097:0:1:869:0:0:0:0:0:0</channel> 
<channel id="409">4097:0:1:192:A:1:3520000:0:0:0</channel> 
<channel id="418">4097:0:1:1258:0:0:0:0:0:0</channel> 
<channel id="326">4097:0:1:982:0:0:0:0:0:0</channel> 
<channel id="524">4097:0:1:1580:0:0:0:0:0:0</channel> 
`

luxorko, 2022-01-23:
[image]

luxorko, 2022-01-23:
would to have option for created "channels ID list" from bouquet or satelite.. like on the attached picture with "EPG id"
hope you know what i mean

DYefremov, 2022-01-23:
Hi,
If you mean something like that: 
[image]

That can be changed to the name of the channel. There is no big difference here. Additional options seem redundant to me.

DYefremov, 2022-01-23:
Changed the numerical value to the name of the channel. 

> hope you know what i mean

As I already wrote about the additional option. It is not entirely clear what it is for and where to place it.

luxorko, 2022-01-23:
would like to get EPG in IPTV channels - coud be simple as possible and quick
scenario:
satelite possition astra 19.2E are scanned; or you have (lamedb file) if cannot be scanned (Astra UK 28E)
all programs-channels had EPG ID in (lamedb)- meaning you can downloading EPG from "epg import plugin" and all channels will showing EPG. does matter you have or not have signal.- we know this
m3u file are transfer to IPTV bouquet list.
IPTV Bouquet channels - if pairing with satelite channels (with same name of course); will transfer EPG ID(from lamedb) to IPTV EPG ID. thad mean all this channels on IPTV Bouquet) will have EPG too.

program could be:
two sides like now 
left side list - have chosee satelite possition, and then channels - right side of choosen iptv bouquet 
left side - showing list of channel with same or simular channel name as on the right side list
then confirm if channel is mach, if not had to showing other possibility (other channel close name) 
after confirm just transfer/add  "EPG ID" to IPTV channel
something user friendly and quick,

luxorko, 2022-01-23:
> Hi,
> If you mean something like that:

could you point me how to get make it again, please.
do option to have epg id from current satelite channels what are had name on it

DYefremov, 2022-01-23:
https://github.com/DYefremov/DemonEditor/issues/51

DYefremov, 2022-01-23:
[Here](https://github.com/DYefremov/DemonEditor/issues/51)

luxorko, 2022-01-23:
have scanned whole orbit is over 12000 channels, 
4000iptv  channels
know this options but is only for manually moving one channel from left side to (choosing) channel on right side
whithout more filtering like by (sattelite,then by channel name, and  authomatic options by program)- imposible to using pairing epg

luxorko, 2022-01-23:
in main is option filter
after click will make more filter setup like this
could you make this for EPG as well please 
[image]
and still prefer to have program choose close chennel matching name automaticly

DYefremov, 2022-01-23:
1. Look closely at this option in the EPG dialog. when this option is activated, the current epg.dat file will be loaded from the receiver and only those channels for which there are EPGs in this file will be displayed in the window on the right.
[image]

 At the moment, this option needs to be worked out to support the newer format (ver 8.) . I am planning in the near future. 
 
2. Channel names in IPTV lists (when importing from m3u) may differ from those in the main (scanned) one. For example, there may be Cyrillic names. Therefore, automatic assignment will not work this way. In addition, you may have duplicate channels (from different satellites but the same name)! What can really be useful is adding a filter by satellite positions to the EPG dialog for the panel on the left. In any case, the issue is not closed and we will need to experiment.

DYefremov, 2022-01-23:
There is also an option similar to yours. It is also possible to load services in the panel on the left only those that are currently displayed in the main list (after the necessary filtering). Needs to experiment as well.

luxorko, 2022-01-23:
dont see any diference if filter by epg.dat activated or vs had channels list not scanned yet.
becouse channels from m3u list (iptv in bouquet) - never same and never know with channels are there. 

best filter i gues could be:
iptv channel on right side - click on it and on the left side pop up same-very close channel name.
then choose (click) on it (left side) and epg assist
very simple might hard to do
anyway i leave it now for your decission and waiting new realese (win10)
all best in new year

DYefremov, 2022-01-23:
Ok. Finally I understand what you want. :) Approximately the same as f[or assigning picons](https://github.com/DYefremov/DemonEditor/issues/49#issuecomment-943432588). I think it's quite feasible.

DYefremov, 2022-01-24:
I've added small improvements to this feature, including auto filtering by name. 

[image]

I hope it will be useful.

luxorko, 2022-01-30:
thank for another great uptade, filter work very vell
in my case (WIN10) bouth channels found (right and left side) then need make "copy refrece" on left side channel, then move to right side channel, click on and make assign
could you find easiet way (just one click) for assign?
like on screen both channels choosing - make pop up icon "assign refferency?" when click on it job done:)

DYefremov, 2022-01-30:
Hi,
> could you find easiet way (just one click) for assign?

Just drag and drop channel from the left side to the desired channel on the right. And there will be exactly one movement! :)

luxorko, 2022-01-30:
in small laptop screen, very hard to do.

DYefremov, 2022-01-30:
Yes, may needs to enable font resizing not only for the main lists but for the rest too.

mapi68, 2022-03-18:
Maybe this is the same problem that I have? DemonEditor after upload settings doesn't write Channel name "<n/a>".
If I zap to channel AFTER write,

[image]
[image]

DYefremov, 2022-03-18:
I don't think it has anything to do with this program. If this is an IPTV list, then it looks more like a WebIf problem (maybe there is a request to lamedb in which this service is missing) or the image itself. If you have any doubts, you can download the complete data set and send me a set of files from the 'data\profile name' folder. I can test.

mapi68, 2022-03-18:
Settings created with DreamEditor and send to STB with DreamEditor ; <n/a>
Settings created with DreamEditor and send to STB with DreamBoxEditor; <n/a>

Settings created with DreamBoxEditor and send to STB with DreamEditor ; <n/a>
**Settings created with DreamBoxEditor and send to STB with DreamBoxEditor ; OK**
[DM7020HD.zip](https://github.com/DYefremov/DemonEditor/files/8307148/DM7020HD.zip)

DYefremov, 2022-03-18:
I don't see any problems on my box.
[image]
[image]

Are you sure you are loading the full dataset, including the lamedb file? 

[image]

DYefremov, 2022-03-18:
If you see N/A in the bouquet for a regular channel, it means that it is missing in the lamedb file or the link to this channel in the bouquet file is incorrectly formed. How were the bouquets created? Are these services (which N/A) displayed in DemonEditor? Do you press 'save' before sending it to the receiver?

mapi68, 2022-03-19:
>>> How were the bouquets created?
DE
>>> Are these services (which N/A) displayed in DemonEditor?
No, only when I transfer to STB
>>> Do you press 'save' before sending it to the receiver?
Yes

Settings created with DreamEditor and send to STB with DreamEditor: <n/a>
Settings created with DreamEditor and send to STB with DreamBoxEditor: <n/a>

Settings created with DreamBoxEditor and send to STB with DreamEditor: <n/a>
Settings created with DreamBoxEditor and send to STB with DreamBoxEditor: OK

DYefremov, 2022-03-19:
> Are these services (which N/A) displayed in DemonEditor?
> No, only when I transfer to STB

This is the key to your problem!  I have a guess. 
If possible, then:
1. Remove all bouquets. 
2. Сreate new one. 
3. Copy 2 channels to a bouquet from the main list. One is **SD** and the other is **HD or UHD**! 
4. Save.
5. Download to the box. 

Do the same in another program (in your case, dreamboxEDIT). For the same channels!  They sent me the settings saved in both programs.

Best

mapi68, 2022-03-19:
Done.
[setting_test.zip](https://github.com/DYefremov/DemonEditor/files/8308849/setting_test.zip)

I think the problem is DE that doesn't write correct transponders section of lamedb. Services section is the same, but DE lamedb is more big than lamedb dbE.
Regards

DYefremov, 2022-03-19:
It's even more interesting. 

> I think the problem is DE that doesn't write correct transponders section of lamedb.

This section is fine. Only the order is slightly different, but this does not affect. 
But there is another interesting point. 
DemonEditor settings:
39d2:00820000:1fa4:013e:1:0:0
Revival TV

dreamboxEDIT settings:
39d2:00820000:1fa4:013e:1:0
Revival TV

We have an extra zero at the end in the DemonEditor settings. If you open in the DemonEditor the settings created by the dreamboxEDIT via File -> Open and then save, the settings (lamedb format) will remain unchanged! Therefore, everything should work. 
Since the DemonEditor tries not to change the format of the channel strings, I would like to understand how the lamedb file was created.

mapi68, 2022-03-19:
Every services in lamedb dreamboxEDIT settings ended with ":0" plus

[image]

DYefremov, 2022-03-19:
I wrote about this above. Try to open the settings without the extra zero through **File -> Open** and hit save!

mapi68, 2022-03-19:
Try this:

1. Open wiith DE
2. Send to STB: it's OK
3. Save file with DE (overwrite)
4. Send to STB: NO WORK <n/a>

[DM7020HD.zip](https://github.com/DYefremov/DemonEditor/files/8309065/DM7020HD.zip)

mapi68, 2022-03-19:
I HAVE FOUND THE PROBLEM!!!!
Simply all files that create (or change) DE is formatted with Windows end line; change to UNIX and it works!

DYefremov, 2022-03-19:
I correctly understood that all this time you used the build for Windows? But I also tried the Windows version. Still need to test again.

mapi68, 2022-03-19:
> I correctly understood that all this time you used the build for Windows? 
Yes.
A small cosmetic thing: why don't add Bouquet tree on the right panel that open when load settings?

[image]

DYefremov, 2022-03-19:
> Yes, I use Windows.

Yes, you were right. Added end of the line in Unix format for lamedb writing. Thanks for your persistence. Somewhat surprised that my experiment with the Windows version is interesting for users of this OS. If it continues like this, then I think that it will be possible to bring it to a completely working state. :)
If there is time, I will try to release a new version in the near future. 

> A small cosmetic thing: why don't add Bouquet tree on the right panel that open when load settings?

Personally, I don't really like this option, but I'll think about it ...

closed 2022-03-20 by DYefremov