DYefremov/DemonEditor #60: control features

odehalaa, 2022-01-25:
First I need to thank you for amazing app and working well with all features and I recommend to add these features if possible please

1 - record control buttons and schedule any time to do any function .
2 - discover all Engima2 boxes on the network.
3- support web interface if possible?

Thanks in advance,

Best Regards

DYefremov, 2022-01-25:
Hello,
It would be nice to clarify some points. 

1. If we are talking about adding a "Record" button to the control panel, then there is no problem. If any other buttons are needed and if there are any possibilities, please describe in more detail. 
2. As I understand it, we are talking about automatic configuration of profiles (auto network search) in the settings dialog. Yes, I already thought about this possibility. Or do you mean something else? 
3. At the moment, for some functions, the program already uses the capabilities of the web interface (HTTP API). Please describe in more detail what you mean!

Best

odehalaa, 2022-01-25:
Hello,
If we are talking about adding a "Record" button to the control panel, then there is no problem. If any other buttons are needed and if there are any possibilities, please describe in more detail.

I mean if we need to record sequence of buttons to do it every specific time or schedule
for example ; 
1- press menu
2- select settings
3- reception
4- automatic scan
5- scan
6- exit

2 - As I understand it, we are talking about automatic configuration of profiles (auto network search) in the settings dialog. Yes, I already thought about this possibility. Or do you mean something else?

auto network search and view all engima2 found it as a list and status active or not.

At the moment, for some functions, the program already uses the capabilities of the web interface (HTTP API). Please describe in more detail what you mean!

I mean if the software can work on like website not application install it on windows
like Linux server and access it from web if possible please?

and I need to thank you again for your kindly support and amazing app

Best Regards

DYefremov, 2022-05-07:
Made a simple version (prototype) of the network explorer.
[image]

odehalaa, 2022-05-09:
very nice feature :), great job

DYefremov, 2022-05-09:
This is just a prototype! If there are comments or suggestions for improvement, I'll be happy to discuss.

DYefremov, 2023-04-15:
> we need to record sequence of buttons to do it every specific time or schedule

I think it would be optimal to implement this feature as a [separate extension](https://github.com/DYefremov/demoneditor-extensions). I have made some experiments. The main issue is scheduling (due to running the app on multiple platforms). But I'll experiment as much as possible. For now, I think this topic can be closed here.

closed 2023-04-15 by DYefremov