DYefremov/DemonEditor #69: iptv stream check

luxorko, 2022-03-05:
he deleted working stream as well, please see bottom examples

`#SERVICE 4097:0:1:A195:B98E:EC:0:0:0:0:https%3a//best-str.umn.cdn.united.cloud/stream?stream=sp1400&sp=n1info&channel=n1bos&u=n1info&p=n1Sh4redSecre7iNf0:N1 B&H
#DESCRIPTION N1 B&H`

DYefremov, 2022-03-05:
Hi,
How this stream was added to the bouquet? 
When adding this stream via "IPTV-> Add IPTV or stream service" everything works as it should. 
Can be verified by adding this initial link: `https://best-str.umn.cdn.united.cloud/stream?stream=sp1400&sp=n1info&channel=n1bos&u=n1info&p=n1Sh4redSecre7iNf0`

[image]

[image]

luxorko, 2022-03-06:
1 - Do confrm stream playing fine in the program.but when during checking stream get removed! 
i mean option: "IPTV - Remove unavailable" when right click on Iptv list. why channel get removed? could fix bug please?
2 - when sending new-update userbouquet old one in stb not removed why? i would like have just new userbouquet and not all together . ist possible change it plase?

DYefremov, 2022-03-06:
1. As already written:

> When adding this stream via "IPTV-> Add IPTV or stream service" everything works as it should.

It has been tested for this function as well. Nothing is deleted! Perhaps there are other examples of similar links? I will check again.
 As well as:

> How this stream was added to the bouquet?

2. Is this option activated in the download dialog? 
[image]

DYefremov, 2022-03-06:
Yes you are right!
There is a problem, but...
Only in **standalone** build for **MS Windows**! There is no problem when running from source.

DYefremov, 2022-03-06:
Point 1.  Yes you are right!
There is a problem, but...
Only in **standalone** build for **MS Windows**! There is no problem when running from source.

luxorko, 2022-03-06:
> 
im using Windows build...
could you make - step by step for dump please?
step 1 download - done
step 2...

DYefremov, 2022-03-06:
Unfortunately, a few words will not do. Perhaps it will be useful to add to the [README](https://github.com/DYefremov/DemonEditor/blob/master/README.md). I'll try to do it if I have free time. For now, I recommend reading about [MSYS2](https://www.msys2.org/). 
Regarding the removal of inaccessible streams, it is relevant starting from version 2.1. Apparently the issue is in a libraries or a packager.

luxorko, 2022-03-06:
> 1. Ako už bolo napísané:
> 
> 
> > Pri pridávaní tohto streamu cez "IPTV-> Pridať IPTV alebo streamovaciu službu" všetko funguje ako má.
> 
> Bol testovaný aj na túto funkciu. Nič nie je vymazané! Možno existujú ďalšie príklady podobných odkazov? ešte skontrolujem. Ako aj:
> 
> > Ako bol tento prúd pridaný do kytice?
> 
>     2. Je táto možnosť aktivovaná v dialógovom okne sťahovania? 
>        [image]

point 2 - thank you didnt see it before

luxorko, 2022-03-06:
> 
will wait for next release

luxorko, 2022-03-06:
> Bohužiaľ, pár slov nepomôže. Možno bude užitočné pridať do [README ](https://github.com/DYefremov/DemonEditor/blob/master/README.md). Pokúsim sa to urobiť, ak budem mať voľný čas. odporúčam prečítať si o [MSYS2 ](https://www.msys2.org/). Čo sa týka odstránenia neprístupných streamov, je relevantné od verzie 2.1. Problém je zrejme v knižniciach alebo v baliarni.

MSYS2 - installed in PC
Demon-Editor 2.2.0 -b1.zip - downloaded
what next?
with "command" should use ? where to put demon.zip file?
 
[image]

DYefremov, 2022-03-06:
Well. You don't give up.  :)

**Step 2:**  Installation of the required packages (for **64-bit**).
`pacman -Suy `
`pacman -S mingw-w64-x86_64-gtk3 mingw-w64-x86_64-python3 mingw-w64-x86_64-python3-gobject mingw-w64-x86_64-python3-requests`

_Optional:_
`pacman -S mingw-w64-x86_64-python3-pillow`

**Step 3:** Unzip the archive folder where MSYS2 is installed.
On my PC it is: 
`C:\msys64\home\test` 

**Step 4:** Run MSYS2 terminal. 
On your screenshot 32 bit version. I am using 64 bit. 
Go to the folder where the program was unpacked.
For example: 
`cd DemonEditor/`
And run:
 `./start.py`

Hope everything works out! :)

luxorko, 2022-03-06:
somthing missing , please advise
[image]

32 vs 64 version - is were installed automatic

DYefremov, 2022-03-06:
Look carefully at steps **2** and **4**. Run a 64-bit terminal version and install the necessary packages.

luxorko, 2022-03-06:
only installed this :
[image]

luxorko, 2022-03-06:
using ming.exe
get error
[image]

DYefremov, 2022-03-06:
Add this line 
`alias python3="C:/msys64/mingw64/bin/python3.exe"`
to the end of the **.bashrc** file as described [here](https://orthodoxpirate.blogspot.com/2020/12/msys2-python-command-not-found-how-to.html).

And instead of command `./start.py ` 
try `python3 ./start.py`

luxorko, 2022-03-06:
sorry not  work
[image]
[image]

DYefremov, 2022-03-07:
Very interesting. It looks like an encoding problem. It looks like the MSYS2 shell is using an encoding other than UTF-8. Perhaps the problem is this. Try to completely reinstall MSYS2. Run **mingw64** shell and install the packages as described above. Everything work for me from the box  without the need to write aliases for Python.
Perhaps [this](https://github.com/msys2/MSYS2-packages/issues/1974), [this](https://stackoverflow.com/a/34108903) or [this](https://www.geeksforgeeks.org/how-to-change-or-set-system-locales-in-linux/) will be useful.
In addition, you can try replacing the _start.py_ file in the source folder with [this](https://github.com/DYefremov/DemonEditor/blob/master/build/win/start.py) one.
Unfortunately I don't have any more ideas.

DYefremov, 2022-03-20:
Should be fixed in a new build.

closed 2022-03-20 by DYefremov

luxorko, 2022-04-14:
thank for hard work 

1 -this [https://zdf-hls-11.akamaized.net/hls/live/2016436/de/high/master.m3u8](url)
is old stream, was working during winter OH. internal player correct detecting stream not work, but when using option "IPTV - Remove all unavaible" channel/stream not deleting and stay in bouquet. 

2 -option in bouquet -"Edit" had bug, sometimes not opening, need change-klick on other channel, come back and then will get open 

3 -OPTION - Bouquet - IPTV - would be nice to had two extra cullum: sort "by URL" and "Bouquet name" where is current channels stored.
reason - comparing other IPTV bouquet (downloaded from net) and deleting channels with same url 

4 - regardning "mingw642" - now giving up, problem between keybord and chair :)

DYefremov, 2022-04-14:
Hi,
1. This function tests the availability of streams. If you do not have access to this stream (e.g. certain authentication), then this **does not mean** that the stream is not available or not working!
2. Need to test.
3. I didn't quite understand where this option should be and how it should look. Please create a separate issue and describe in more detail.
4. It is also not entirely clear what is meant. Perhaps it makes sense to ask a question [here](https://github.com/msys2).

luxorko, 2022-04-14:
stream test - point 1 - OK, i think if picture or sound are not avaible - stream is dead and should be deleting. could you make aware after checking with channels-stream are not transmiting (no picture) - and will be option for deleting manually?

luxorko, 2022-04-14:
4- strugling make compilaton from source code "DemonEditor". dont mind . will use and wait for your new compilation on github

luxorko, 2022-04-14:
OK