DYefremov/DemonEditor #81: Multiple bouquets

i-mirror, 2022-03-21:
It is possible to import multiple bouquets at the same time?

[image]

DYefremov, 2022-03-21:
Yes, sure.
[image]

> When using the multiple import feature, from lamedb will be taken data only for channels that are in the selected bouquets!

i-mirror, 2022-03-21:
I only want to import previously saved bouquets, without services.

DYefremov, 2022-03-21:
If services are already in the main list, then nothing will be imported additionally. Just bouquets only.

i-mirror, 2022-03-21:
OK, thanks.

DYefremov, 2022-03-21:
Any time.

closed 2022-03-21 by DYefremov