DYefremov/DemonEditor #87: Build instruction on Windows.

DYefremov, 2022-04-01:
It is necessary to create a separate instruction for launching and building a standalone package in MS Windows.

mapi68, 2022-04-01:
Can you explain more verbose?
Started from [here](https://github.com/DYefremov/DemonEditor/tree/development-2.0/build/win) with .spec file and pyinstaller.....

DYefremov, 2022-04-02:
## Launch
The best way to run this program from source is using of [MSYS2](https://www.msys2.org/) platform. 
1. Download and install the platform as described [here](https://www.msys2.org/) up to point 4. 
2. Launch **mingw64** shell. 
[image]
3. Run first  `pacman -Suy ` After that, you may need to restart the terminal and re-run the update command. 
4. Install minimal required packages: `pacman -S mingw-w64-x86_64-gtk3 mingw-w64-x86_64-python3 mingw-w64-x86_64-python3-gobject mingw-w64-x86_64-python3-pip mingw-w64-x86_64-python3-requests`
Optional: `pacman -S mingw-w64-x86_64-python3-pillow`
To support streams playback, install the following packages (the list may not be complete): 
For [MPV](https://mpv.io/) `pacman -S mingw-w64-x86_64-mpv` , 
For  [GStreamer](https://gstreamer.freedesktop.org/)  `pacman -S mingw-w64-x86_64-gst-libav mingw-w64-x86_64-gst-plugins-bad mingw-w64-x86_64-gst-plugins-base mingw-w64-x86_64-gst-plugins-good mingw-w64-x86_64-gstreamer`
5. Download and unzip the archive with sources from preferred branch (e.g. [master](https://github.com/DYefremov/DemonEditor/archive/refs/heads/master.zip)) in to folder where MSYS2 is installed. E.g: `c:\msys64\home\username\`
6. Run mingw64 shell. Go to the folder where the program was unpacked. E.g: `cd DemonEditor/`
And run: `./start.py`

## To be continued...

DYefremov, 2022-04-03:
## Building a package
To build a standalone package, we can use [PyInstaller](https://pyinstaller.readthedocs.io/en/stable/). 
1. Launch mingw64 shell.
2. Install PyInstaller via pip:  `pip3 install pyinstaller`
3. Go to the folder where the program was unpacked. E.g: `c:\msys64\home\username\DemonEditor\`
4. Сopy and replace the files from the /build/win/ folder to the root .
5. Go to the folder with the program in the running terminal:  `cd DemonEditor/`
6. Give the following command: `pyinstaller.exe DemonEditor.spec`
7. Wait until the operation end. In the dist folder you will find a ready-made build.

### Appearance
To change the look we can use third party [Gtk3 themes and Icon sets](https://www.gnome-look.org) . 
To set the default theme:
1. Сreate a folder "`\etc\gtk-3.0\`" in the root of the finished build folder.
2. Create a _settings.ini_ file in this folder with the following content: 
  ```
  [Settings]
  gtk-icon-theme-name = Adwaita
  gtk-theme-name = Windows-10
  ```
In this case, we are using the default icon theme "Adwaita" and the [third party theme](https://github.com/B00merang-Project/Windows-10) "Windows-10".
Themes and icon sets should be located in the `share\themes` and `share\icons` folders respectively. 
You can find more info about changing the appearance of Gtk applications on the Web yourself. 
### Corrections and additions are welcome!

luxorko, 2022-04-06:
get error ..
`My@LAPTOP-OM27EOQ1 MINGW64 ~/Demon
$ ./start.py
Traceback (most recent call last):
  File "C:\msys64-\home\My\Demon\start.py", line 26, in <module>
    from app.ui.main import start_app
  File "C:/msys64-\home\My\Demon/app/ui/main.py", line 38, in <module>
    from gi.repository import GLib, Gio, GObject
ModuleNotFoundError: No module named 'gi'`

DYefremov, 2022-04-06:
I think not all dependencies were installed, or not installed correctly. In your case it looks like the package '`mingw-w64-x86_64-python3-gobject`' is missing.

luxorko, 2022-04-06:
after re-install get this:
`My@LAPTOP-OM27EOQ1 MINGW64 /home/demon
$ ./start.py
Traceback (most recent call last):
  File "C:\msys64\home\demon\start.py", line 28, in <module>
    update_icon()
  File "C:\msys64\home\demon\start.py", line 10, in update_icon
    lines = f.readlines()
  File "C:/msys64/mingw64/lib/python3.9/encodings/cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 129: character maps to <undefined>
`

DYefremov, 2022-04-06:
I already answered [here](https://github.com/DYefremov/DemonEditor/issues/69#issuecomment-1060813386). Unfortunately, I can't add anything new. Personally, I can't reproduce it. Perhaps for testing purposes, it makes sense to try changing the locale of the system itself. For example in English.

luxorko, 2022-04-07:
OS Win 10- EN

`My@LAPTOP-OM27EOQ1 MINGW64 ~
$ locale
LANG=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="C.UTF-8"
LC_TIME="C.UTF-8"
LC_COLLATE="C.UTF-8"
LC_MONETARY="C.UTF-8"
LC_MESSAGES="C.UTF-8"
LC_ALL=
`

DYefremov, 2022-04-07:
Yes, I can confirm. Really interesting. Occurs on Win 10 if the system locale is `en_US`.

> In addition, you can try replacing the start.py file in the source folder with [this](https://github.com/DYefremov/DemonEditor/blob/master/build/win/start.py) one.

This should help! In addition, I slightly changed the main start script. It should also work, but I still recommend the method above!