DYefremov/DemonEditor #91: Are you sure? OK / Cancel in Windows and Linux

mapi68, 2022-04-13:
Tried version 2.2.3 with Raspberry Pi (debian) and it's AWESOME, but there is a difference in Win and Linux: default delection in Win is OK (left) an in Linux is Cancel (always on left). I think that the right way is Windows because you can confirm direct with keyboard ENTER

WIndows
[image]

Linux
[image]

DYefremov, 2022-04-13:
This location depends on the Gtk settings, OS and used desktop environment. This can be changed, but I see no reason to do so.

DYefremov, 2022-04-13:
> No lose time to search OK with mouse and do click (simply ENTER in keyboard).

This is the **default** behavior! I also think this is the correct behavior. I think that a significant part of Linux users think the same.

> Or maybe add something like this:

I'm not sure if it's necessary, but I'll think about it.

closed 2022-11-27 by mapi68