DYefremov/DemonEditor #92: extra cullum

luxorko, 2022-04-14:
OPTION - Bouquet - IPTV - would be nice to had two extra cullum: sort "by URL" and "Bouquet name" where is current channels stored.
reason - comparing other IPTV bouquet (downloaded from net) and deleting channels with same url
[image]

DYefremov, 2022-04-14:
It may be useful, but it can slow down the addition of data. Will need to experiment.:)

DYefremov, 2022-04-27:
I had free time and I did some experiments. 
Since this list primarily serves to order channels between bouquets, and the channel can move and be located in several bouquets at once, tracking the current state can complicate the program logic. 
[image]
So a separate column for the bouquet name does not seem like a good idea to me.
The URL column is more efficient, but again, on large lists, it can slow down the work. Since adding data will require additional post-processing. But this is a working option. In addition, it does not require much effort.

It seems to me that it would be more correct to filter out duplicates at the import stage. Or add a function like searching for unavailable streams. I think there may be more options.

closed 2022-08-22 by DYefremov