DYefremov/DemonEditor #100: [Errorno 2] No such file or directory /home/......./satellites.xml after web import

ahmedmoselhi, 2022-04-26:
i have imported satellites and services from lyngsat but after that i got only services but no satellites.xml file downloaded

DYefremov, 2022-04-26:
This feature only loads channels (updates _lamedb_). To load and update the list of satellites, there is a separate tab with  the upde-edit satellites ( -> _satellites.xml_) feature.

[image]

ahmedmoselhi, 2022-04-26:
> This feature only loads channels (updates _lamedb_). To load and update the list of satellites, there is a separate tab with the update-edit satellites ( -> _satellites.xml_) feature.
> 
> [image]

ok i found it .. one more thing is there a plan to add comparision between two different channel lists in future releases?

DYefremov, 2022-04-27:
No. Such functions were not planned. But...
For **point 1**, you can create a separate issue with a description of what and how it should look and work. We need to understand what and how you want to compare.
For **point 2**. The purpose of this division is not entirely clear. In addition, if we are talking about satellite update from the Web, then the structure is provided by the site itself. Yes, we can do post-processing, but as I wrote above, it is not entirely clear to whom and why this is needed. Personally, I see it as redundant. Аlthough it is certainly possible.

closed 2022-04-28 by DYefremov