DYefremov/DemonEditor #101: Add Send / Receive Picons from FTP transfer window

mapi68, 2022-04-28:
This is an example what I mean.....

[image]

DYefremov, 2022-04-28:
Yes, there are no problems with this. I can add it. But the Send/Receive buttons on the 'Picons' tab play another role also. You can drag a picon from the explorer to the desired button and, depending on the button, the piсon will be uploaded or downloaded. 
I plan to do some refactoring. And perhaps this dialog will not exist at all. There will be buttons common to all tabs on the toolbar next to the profile selection. And depending on the tab, the necessary data will be loaded. But I haven't decided yet... :) There may be other options...

DYefremov, 2022-04-28:
Approximately as in this program. 

[image]

This is another  [project ](https://github.com/DYefremov/E2Toolkit)from me.

DYefremov, 2022-04-29:
At the moment i have added your variant. I hope it will be helpful.

[image]

mapi68, 2022-04-30:
It works for Picons only if it's selected "Picons", but with "All" only take Bouquets + Satellites.
Isn't it better remove radio buttons and put check?

[image]

DYefremov, 2022-04-30:
> It works for Picons only if it's selected "Picons", but with "All" only take Bouquets + Satellites.

Yes, that's how it was meant to be.

> Isn't it better remove radio buttons and put check?

Yes, this is also a working option. The main question is what is more convenient. Need to experiment.

This option also needs to be considered.

> There will be buttons common to all tabs on the toolbar next to the profile selection. More or less like this:
> 
> [image]

 I think that would be even more convenient. But this is just a guess. Users may think differently.

mapi68, 2022-04-30:
For me it will be beautilful with buttons common to all tabs on the toolbar next to the profile selection.

Please consider to add in Settings ---> Program "Compress PIcons before send to receiver"
You the Best

DYefremov, 2022-04-30:
> For me it will be beautilful with buttons common to all tabs on the toolbar next to the profile selection.

Ok. For now, let's stick with this option. But this will require significant changes in the app. In any case, I've already planned something similar. :)

> Please consider to add in Settings ---> Program ---> "Compress PIcons before send to receiver"

I don’t see any big problems here, but I need an exact understanding of what is meant by:

> Compress PIcons

it is not entirely clear what or how compression should occur.

mapi68, 2022-04-30:
>> Compress PIcons

> It is not entirely clear what or how compression should occur.

If you have 2 or more satellites it's normal to have more than 5000 picons and transfer with FTP may will be slow. 
To speedup the process DE can compress ALL PICONS with 7z or tar.gz, send the archive to receiver in /tmp/ and after unzip it to picons folder of receiver.

DYefremov, 2022-04-30:
Unfortunately, this doesn't make much sense.
1. Just compress one *.png file and compare the size before and after.
2. Add time to compress and then decompress. If this is a modern receiver, it may not be so critical.
3. How do you plan to unpack and then move (what commands, plugins, scripts) files?

mapi68, 2022-04-30:
> Unfortunately, this doesn't make much sense.

dreamboxEDIT make this
> How do you plan to unpack and then move (what commands, plugins, scripts) files?

**This is the log of dreamboxEDIT** 
Compressing Picons
** Sending file "/tmp/picon.zip" **
FTP-Transfer wird gestartet
Transfer abgeschlossen
Verbindung wird getrennt.
Verbindung getrennt.
Sending Reload commands.
Reload of lamedb and userbouquets succeded.
Reload of satellites.xml succeded.
Reload of blacklist succeded.
Telnet status: Host-Name dm900 wird aufgelöst.
Telnet status: Verbinden mit 192.168.1.20.
Telnet status: Verbunden.
Sending username root
Sending password ********
Sending username root
Extracting Picons
Picons successfully extracted
Terminating telnet
Telnet status: Verbindung getrennt.

DYefremov, 2022-04-30:
Yes I know that. :) Telnet option. But I refused this opportunity at the design stage. Unfortunately I don't remember why. Perhaps there were problems with unpacking (for Boxes  with a small amount of memory. As I already wrote, for modern receivers, apparently no longer relevant) or with the library used for telnet. Of course, nothing prevents us from trying again, as an option in the form of an experimental feature that is separately enabled by the user.

mapi68, 2022-04-30:
Yes, problems with unpacking for low memory decoder.
But if you have in Settings ---> Program ---> "Compress PIcons before send to receiver" disabled you send picons one by one and don't have this problem.

DYefremov, 2022-04-30:
Ok.
> nothing prevents us from trying again

We can experiment.:) But by default, this option should be disabled.

DYefremov, 2022-05-04:
I have added automatic picon compression during upload if the selected picon count > 1000. 
[image]

I think this would be a compromise option.

mapi68, 2022-05-04:
Yes, but for oldest receiver with low memory it's better to add a check of picons.zip size if is greater than free space in /tmp/

DYefremov, 2022-05-04:
Added option to settings:
[image]

mapi68, 2022-05-05:
Works good with compression and transfer to receiver, but after when receiver with telnet extract, it stops after transfer 100-200 picon of 4000 and remove picons.zip
Maybe telnet timeout problem?
Tried to change  up timeout of telnet in connections.py and extract more icons.

DYefremov, 2022-05-05:
My mistake. Test again!

closed 2022-05-17 by DYefremov