DYefremov/DemonEditor #104: subbouquet.******.tv

angelheart150, 2022-05-05:
dears 
I have an issue to read file started with (**subbouquet****.****.** ) from E-channelizer app
DemonEditon  didn't read it 

please solve this issue 

thanks

DYefremov, 2022-05-05:
Are you sure the files are created correctly? I don't have this program. It would be nice to have bouquet files for testing.

DYefremov, 2022-05-05:
I have tested your settings. Can you clarify how the bouquets that are not visible were added (subbouquet.*)? Your bouquets.tv file contains links to sub-bouquets. Which is strange and not logical. Usually this file contains links to user bouquets (userbouquet) which may already contain links to sub-bouquets (subbouquet).

[image]

Yes, I can add support for reading such configurations, but when saving, the sub-bouquet will be (converted) saved as a "userbouquet.*" one.

angelheart150, 2022-05-05:
I know It will appear when saving in the backup folder with any error when we make it from demoneditor 
But my error happened when i add it from external program , so i think the problem in my files and lamdb the reason of that.
I think i will build my channel files from the beginning by demon editor it will do it with success 
But i need to know why i can't do multi selection in bouquets box ?

DYefremov, 2022-05-05:
Ready. Please test! 

[image]

I hope I didn't break anything.:)

> But i need to know why i can't do multi selection in bouquets box ?

I don't quite understand what you mean. Can you explain in more detail.

angelheart150, 2022-05-05:
> Ready. Please test! 
> 
> [image]
> 
> I hope I didn't break anything.:)
> 
> > But i need to know why i can't do multi selection in bouquets box ?
> 
> I don't quite understand what you mean. Can you explain in more detail.
> 

I understand now 
I don't like sub-bouquets and by e-channelizer i had moved sub-bouquets to normal bouquets by copy and paste
E-channelizer treat with it as i copied normal because this data saved in lamdb  but when i used demon it wasn't read it as normal bouquets for reason i don't know .

DYefremov, 2022-05-05:
If we are talking about a list of bouquets, then press the CTRL key and select the desired bouquets with the mouse. If you need to move them then holding CTRL move the up and down arrows. Or just drag and drop. If you need to copy the contents of the bouquet or move it, then as in any editor ( CTRL + A, CTRL + C /X/ V). Or through the context menu.

angelheart150, 2022-05-05:
> If we are talking about a list of bouquets, then press the CTRL key and select the desired bouquets with the mouse. If you need to move them then holding CTRL move the up and down arrows. Or just [...]

Every thing working now without any issue thanks alot .

DYefremov, 2022-05-06:
Well. Then closed.

closed 2022-05-06 by DYefremov