DYefremov/DemonEditor #115: browsing speed

luxorko, 2022-08-21:
first downloading bouquet from STB, movement in channels list are very slow, this will gone once you save bouquet to (PC) and opening again

- might program save bouquet to PC and opened automatic after downloading from stb? (folder name - date+time?)

DYefremov, 2022-08-21:
> first downloading bouquet from STB, movement in channels list are very slow, this will gone once you save bouquet to (PC) and opening again

Possibly related to the OS used and library versions. Need to test. What OS version are you using?

> * might program save bouquet to PC and opened automatic after downloading from stb? (folder name - date+time?)

It's not entirely clear what you mean.

luxorko, 2022-08-21:
- OS -Win 10

DYefremov, 2022-09-26:
Please test with [this version](https://github.com/DYefremov/DemonEditor/releases/tag/3.0.0-a2).

luxorko, 2022-10-01:
without any lag.smooth movement in all menu , excellent work thank you

DYefremov, 2022-10-02:
Ok. We can close.

closed 2022-10-02 by DYefremov