DYefremov/DemonEditor #117: Picons assign

luxorko, 2022-08-21:
option picon - when click on channel, channel name should by automatic showing in "filter" and showing all availables picons

filtering - if channel name is (205_Boomerang) ; nothing find becouse he looking for whole channel name , but when just wrote "boomerang" channels find it. could be improve filter - filtering? 

In Picon card (right on picture - could you add "sat. possition + provider+ frequence) - this will really help to choosing right picon for iptv channel  

when assign picons - picon references should be transfer as well for channel as well. otherwise all channels with same reference get same picon.
 
[image]

DYefremov, 2022-08-21:
Hi,
> option picon - when click on channel, channel name should by automatic showing in "filter" and showing all availables picons

The "auto" switch must be enabled.

> when assign picons - picon references should be transfer as well for channel as well. otherwise all channels with same reference get same picon.

If the channels have the same link, then it is the same channel for Enigma2! 

> In Picon card (right on picture - could you add "sat. possition + provider+ frequence) - this will really help to choosing right picon for iptv channel

[image]

> filtering - if channel name is (205_Boomerang) ; nothing find becouse he looking for whole channel name , but when just wrote "boomerang" channels find it. could be improve filter - filtering?

I can't reproduce it. Needs to test.

luxorko, 2022-08-21:
The "auto" switch must be enabled. - picons not found it 
[image]

If the channels have the same link, then it is the same channel for Enigma2

ALL IPTV channels have same basic reference
 
[image]

could you make could be possible - during assign picons to iptv channel will be "change - rewrite " picon reference to iptv channel??

DYefremov, 2022-08-22:
> during assign picons to iptv channel will be "change - rewrite " picon reference to iptv channe

Everything you need is already on board. You can configure the IPTV list when importing into a bouquet, or after (IPTV menu -> List configuration). 

[image]

> ALL IPTV channels have same basic reference

See above.

luxorko, 2022-08-22:
> Everything you need is already on board.
sorry, in win 10 looks diference..
[image]

scenario:
- we have IPTV list for" Germany channels". we know this channels are on "Astra 19.2"  
picon card will download all picons from 19.2
[image]

- than go to picon card "explorer card" and choose right picon for IPTV channel (Boomerang) 
the card "explorer" should have more column (sat.possition + provider + freq) -helping to choosing right picon
[image]

- now just double click on picon and prog."demoneditor" will ask "would you like to assign picon for (channels name) with reference?" 
click yes and job done. 
nice and easy
could by possible? what do you think?

l

DYefremov, 2022-08-22:
> sorry, in win 10 looks diference

My dialog screenshot is for m3u import.

> * now just double click on picon and prog."demoneditor" will ask "would you like to assign picon for (channels name) with reference?"

You can select several channels in the bouquet and drag the piсon from the explorer to the selected services in the bouquet! Try it!

> * the card "explorer" should have more column (sat.possition + provider + freq) -helping to choosing right picon

I think it can be made. For example like here:
[image]

Or with additional column with extra info.

luxorko, 2022-08-22:
> You can select several channels in the bouquet and [drag](url) the piсon from the explorer to the selected services in the bouquet!

closed 2022-08-22 by luxorko

luxorko, 2022-08-22:
> filtering - if channel name is (205_Boomerang) ; nothing find becouse he looking for whole channel name , but when just wrote "boomerang" channels find it. could be improve filter - filtering?
> 
 could you update please

luxorko, 2022-08-22:
> > filtering - if channel name is (205_Boomerang) ; nothing find becouse he looking for whole channel name , but when just wrote "boomerang" channels find it. could be improve filter - filtering?
> 
> I can't reproduce it. Needs to test.

in "filter " just add some number before channel name :)

DYefremov, 2022-08-22:
> I can't reproduce it. Needs to test.

[image]

DYefremov, 2022-08-22:
> in "filter " just add some number before channel name

Then it's your fault. Are you looking for 'google' or '1google'?

DYefremov, 2022-08-22:
> yes is work but problem are reference...all IPTV in bouquet have same reference

> can configure the IPTV list when importing into a bouquet, or after (IPTV menu -> List configuration).