DYefremov/DemonEditor #121: saving folder

luxorko, 2022-08-23:
- downloading bouquet from STB to program
- make some changed and bouquet save as to folder "23.08"
- opening program demon - opening buquets from folder "23.08"
- do same change and press " SAVE" in File 

except to be save it there, where get opening from, BUT saving do not save in folder 23.08
all get save to path on picture
[image]

luxorko, 2022-08-23:
no much user friendlly. Thank you letting us know. 
ill close this now

closed 2022-08-23 by luxorko

DYefremov, 2022-08-23:
Perhaps not usual, but quite justified. For example, you do not change the external source data (keep it intact). If you really need to change the given source, just use "Save As".  It's more a matter of preference... :)