DYefremov/DemonEditor #122: filter result

luxorko, 2022-08-23:
have fiter word " RTL" and result are bit stange, why no by alphabet? ... see picture  

[image]

filtering i mensioned before (picon; EPG...)
any change for upgade, please

DYefremov, 2022-08-23:
> why no by alphabet?

Because it is a sorting function and not a filtering function. The gas pedal is not for slow down either.

luxorko, 2022-08-23:
im just user,...thank for explain.
now is closed

closed 2022-08-23 by luxorko

DYefremov, 2022-08-23:
You should understand that when filtering the data you expect to get some set. But it is not at all obvious that users also need to sort this data. For example, it is not clear to sort by which column, in what order, etc. In addition, sorting at the time of filtering will also slow down the process of filtering itself. Sometimes very significant. I think this is not wat users need.