DYefremov/DemonEditor #123: problem with paths

luxorko, 2022-08-25:
change patch - choosing folder - save - asking choose folder - save - cyckling question
only close program help keep patch
[image]

file - opening -going to (see picture) = should search folder by "patch"    
[image]

DYefremov, 2022-08-26:
Unfortunately, I can't reproduce it. Works on all my OSes. Please specify your System. OS -> Version -> Language  - etc. I'll try to test it in a virtual machine.

luxorko, 2022-08-26:
win 10 home - eng

luxorko, 2022-10-04:
problem remain in  3.0.0 beta

DYefremov, 2022-10-05:
This problem has not yet been resolved. Perhaps it is related to the peculiarities of the work or the bug in the graphic library on Windows. I found a solution to this problem, but it requires changing the widget type (for example, as in the 'Picons' -> 'Converter' tab), which will also affect other systems, which I don't want yet. Although the option is quite working. I'm just not completely sure that it will fit in this case.

luxorko, 2022-10-05:
Although the option is quite working. I'm just not completely sure that it will fit in this case.

i could test for you

DYefremov, 2022-11-12:
Please test [this version](https://github.com/DYefremov/DemonEditor/releases/tag/3.1.0-a1)!

DYefremov, 2022-11-13:
https://github.com/DYefremov/DemonEditor/issues/130#issuecomment-1312592010

closed 2022-11-13 by DYefremov