DYefremov/DemonEditor #125: debian: archive 'DemonEditor-3.0.0-Alpha2.deb' uses unknown compression for member 'control.tar.zst', giving up

franck-888, 2022-10-01:
FYI on Debian GNU/Linux 10 (buster):
• DemonEditor-3.0.0-Alpha2.deb uses an unknown .deb format.
• DemonEditor-2.2.4-Beta.deb  works well.
See below for details.
BTW: Thx for this very handy tool :)

dpkg -i DemonEditor-3.0.0-Alpha2.deb 
dpkg-deb: error: archive 'DemonEditor-3.0.0-Alpha2.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive DemonEditor-3.0.0-Alpha2.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 DemonEditor-3.0.0-Alpha2.deb

dpkg -i DemonEditor-2.2.4-Beta.deb 
(Reading database ... 130956 files and directories currently installed.)
Preparing to unpack DemonEditor-2.2.4-Beta.deb ...
Unpacking demon-editor (2.2.4-Beta) over (2.2.3-Beta) ...
Setting up demon-editor (2.2.4-Beta) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...

DYefremov, 2022-10-01:
Hi,
Yes, it looks like after upgrading my distro dpkg is using a different default compression type that is [not supported by Debian](https://unix.stackexchange.com/questions/669004/zst-compression-not-supported-by-apt-dpkg). I have added compression option to the build script. 
Please test this package:
[DemonEditor_3.0.0_Alpha2.deb.zip](https://github.com/DYefremov/DemonEditor/files/9690669/DemonEditor_3.0.0_Alpha2.deb.zip)

franck-888, 2022-10-01:
Yes it works = I can install your package.
Thx for your fast reaction.

DYefremov, 2022-10-02:
Fine! We can close.

closed 2022-10-02 by DYefremov