DYefremov/DemonEditor #126: service dvb search

luxorko, 2022-10-04:
1- when searching word "cinema" -  find it first channel with word "cinema" in the name -
 better will be if on first possition will be showing all channels beginning with word "cinema" and then will be showing channels where word "cinema" on the other possition in channel name

2 - please see same picture - in column service - channels name get cutting name early even space there "150 SuperNova Trze...."

[image]

DYefremov, 2022-10-05:
p. 1. I don't think that's a good idea. Since additional data processing is required (sorting + moving). This can significantly slow down the search.
With regards to the p. 2. Ill test what you mean. But I think, it should be all ok here. I'll test.

luxorko, 2022-11-12:
With regards to the p. 2. Ill test what you mean. But I think, it should be all ok here. I'll test.

version 3.1.0 still problem with shrinking "channel name in DVB bouquet

[image]

DYefremov, 2022-11-13:
Thanks to your screenshot, I finally understand what you mean. :) The ''Service'' column has a limit on the number of displayed characters  (< 25). In the next version I will try to increase this value or even remove the limit.

luxorko, 2022-12-06:
in the "Service" column showing full channel name. great job.

Could by also possible to make "reverse" please - channels name are srink - cutting name early please see attached pictute
channel name "Eleven Sports Pro League 1 NL" - only two character "NL" missing - should be vissible without srink namw 

[image]

DYefremov, 2022-12-06:
> lots af space in culumn avaiable

Since this place can be used for other things.

[image]

closed 2022-12-24 by DYefremov