DYefremov/DemonEditor #127: import service detail

luxorko, 2022-10-04:
could be possible to add option to choose with service will be importing ?
now we can choose just from bouquet list, ,,,,and all servise will be importing anyway

[image]

DYefremov, 2022-10-05:
Yes. I think it's possible.

DYefremov, 2022-10-05:
I have added  basic implementation of this feature.
[image]

luxorko, 2022-11-12:
still missing option import "just by services" like (omport setting by satelite possition 19.2E or 30w)

DYefremov, 2022-11-13:
Then **I don't understand** what you need. You can choose a satellite and  services which you need.

[image]

luxorko, 2022-11-13:
demon have fantastick function joint two enigma settings ( two lame db) together...
simply import channels (from lamedb) "from  Astra 28E (lamedb)" to current enigma setting

hope is clear

verion 3.0.0 have this future, if you import bouquet.tv (bouquet contains channels from Astra 28E) then will importing all channels from (LAMEDB Astra 28E) to current enigma2 setting (lamedb)

DYefremov, 2022-11-13:
> could be possible to **add option to choose with service will be importing** ?
> now we can choose just from bouquet list, ,,,,and all servise will be importing anyway

I have added **what you asked**!

> simply import channels (from lamedb) "from Astra 28E (lamedb)" to current enigma setting = will rewrite lamedb - Astra 28E
>from current setting and add new lamedb Astra28E from importing list

Just create a new configuration -> Import new settings -> Import old ones. And you will have updated settings. Or do you need an option that allows you to overwrite the current data? Then it's not this function:

> option to choose with service will be importing 

but quite different. Please describe clearly what you want. Step by step. If you need **grouping by satellites**, then this is also another function!

luxorko, 2022-11-15:
a - i have own enigma2 setting - i scanning orbit myself.
b - downloading enigma2 setting (all orbit) and would like to import ONLY Astra 28.2E

Step by step new function
File - Import -(choose enigma2setting) - show all available satellites: Astra 19.2, Hotbird 13, Astra 28.2 ....
- choose to import with satellite possition will be adding; etc ASTRA 28.2 (without bouque)  
- rewriting if same satellite possition exist (delete old - add new)
thank you

PS. satellite possitions=scanned channels=services=lamedb

DYefremov, 2022-11-15:
OK. Now it's clearer. :) In short, we need a **grouping by satellites**. I think we can add an extra tab for this feature. I think we cat try to implementing this in the next version.

luxorko, 2022-11-25:
did you add this option? v3.1.0 beta not see it

DYefremov, 2022-11-25:
No. 

> We can try to implement this in the next version.

This requires quite a lot of work.

luxorko, 2022-12-11:
excelent.
Waiting for win10 build. 
ps dont forget add 7zip :)

luxorko, 2022-12-25:
great! import by satellite possition working.
no sure if possible to delete same channels after importing new enigma2 setup.

meaning: currentlly have channels 13E. if do importing again same channels from 13E then demon will importing all channels, but some channels allready exist in both setup "current and new" (they have same name , same frequency, same pids...) could by possible for demon just import only new channels (they not exist in current setup) ? now are all channels twice plus 2-3 new channels

DYefremov, 2022-12-25:
Ok. Corrected!

luxorko, 2023-01-04:
new channel name are not updated-imported

could you make test pleas:
-import Vhanibal setting before 20 dec 22 
[https://www.linuxsat-support.com/thread/151841-vhannibal-enigma2-setting/?pageNo=8](url)

then import current Vhaniball setting from 2 Jan 23  
[https://www.linuxsat-support.com/thread/151841-vhannibal-enigma2-setting/?pageNo=9](url)

on Astra 23.5 started 4 new Serbian channels (from 24 Dec) Eurocinema 1;2;3 and 4.

after import new setting: will change only picons but not channel name stay old

  
[image]

DYefremov, 2023-01-04:
It's all right here. At the moment the import works as before. If reference of the imported service matches one of the channels in the main list, then such channel is skipped.In the future, an additional option is planned to allow replacement.
E.g like this:
[image]

luxorko, 2023-01-20:
version 3.2.3Beta dont have this future " replace existing"

DYefremov, 2023-01-20:
No. The current update contains bug fixes and minor improvements only. This feature requires more significant changes in the code. This is planned in the next major (e.g 3.3.0) version.

luxorko, 2023-02-26:
current version 3.4.2 - working OK 
thank you

closed 2023-02-26 by DYefremov