DYefremov/DemonEditor #128: picon enhanced

luxorko, 2022-10-08:
could be possible to have support for Picons by 'Service name" please

DYefremov, 2022-10-09:
If for display in the program, then I do not see any big problems. Main question is assignment. How to combine the two variants. Here needs to think. One global option or, maybe, some additional menu items.

mapi68, 2022-10-10:
Could be possible to have support for DTT Italy picons.cz?
With condition If exists "DIgitale Terrestre" in bouquet name....

DYefremov, 2022-10-10:
Did I understand correctly that this combination does not work?
[image]
If so, it is desirable to send me your set of channels.

> With condition If exists "DIgitale Terrestre" in bouquet name....

Of course it can be made, but it's a bad idea (if I understand correctly). Such a feature should be universal, and not tied to a specific provider or name. Perhaps we need some additional input field or something like that.

mapi68, 2022-10-10:
It works, I had filter enabled.
Is it possible if vHannibal settings SAT + DTT are installed that works with filter enabled?
[image]

DYefremov, 2022-10-10:
Since the filtering occurs by position, and for these packages the author of the picons does not provide such information, it is currently impossible.

DYefremov, 2022-10-10:
Since the filtering occurs by position, and for these packages the author of the picons does not provide such information (API peculiarity), it is currently impossible.

chocholousek, 2022-11-05:
> Since the filtering occurs by position, and for these packages the author of the picons does not provide such information (API peculiarity), it is currently impossible.

Hello, what exactly does the DE look for in the package name for DTT?

DYefremov, 2022-11-05:
Hi,
When parsing the mapping file, we have, for example, such results:
1. '8178' ->  'piconmirrorglass-400x240-**DTTitaly**_by_chocholousek.7z'
2. '8166' ->  'piconmirrorglass-400x240-**4.8E**_by_chocholousek.7z'

DE parses the file name and looks for the position (**4.8E** in our case). But there is no position for the value 'DTTitaly'!
[image]

 Also, this value is not in other columns of the main channel list (for example, we could use provider name etc.)
Filtering and searching through bouquets list does not make sense. Since users can use any names and bouquets set. Perhaps there are other options. Ready to discuss.

Anyway, thanks for your picon sets. They are really cool!

chocholousek, 2022-11-06:
Hello, and thanks. ;)
I understand, I thought that DE is looking for some nickname dvb-t dtt or terr for terrestrial as a replacement and non-existent position. But it's better to leave everything as it is.

DYefremov, 2022-11-11:
> could be possible to have support for Picons by 'Service name" please

Added basic support. So far, only for display in the program. As for the assignment, it is not yet entirely clear how best to do this.

chocholousek, 2022-11-12:
Possible but not easy. The whole project is adapted only for service reference. I'll try to figure something out.

DYefremov, 2022-11-12:
Of course it's possible. :) The main question is how to make it more user-friendly. Additional menu item, additional function or option for conversion (for example, on the "Picons" tab), etc.

chocholousek, 2022-11-20:
Can you explain that?

DYefremov, 2022-11-20:
If short...
You have channel names:
1. HTB.
2. НТВ.

They looking the same. But... These are different names.
1 - written Latin alphabet, and 2 nusing the Cyrillic alphabet!
[Here is a function](https://github.com/DYefremov/DemonEditor/blob/c5c482353402738392b5ab8738665af427faabda/app/ui/main_helper.py#L568) that converts the service name to the necessary name of a picon. Everything seems to be logical and simple.  BUT the ASCII table does not contain all characters used by Cyrillic! We need transliteration!  That is, before we convert the name, we need to find out in what language and in what encoding it is written, and then convert it according to certain rules! Since the lists can be large and contain names in different languages, the performance issue can become quite really.

chocholousek, 2022-11-20:
Yes, I also encounter this problem in the Czech language. It's been a long time since I created my own script that solves this problem.
[image]
A problem that can be solved.

DYefremov, 2022-11-20:
> A problem that can be solved.

Yes, in your particular case. But keep in mind that the terminal also uses certain language settings for each specific system. Also, when run from Python, the result may be different. In addition, you have an example for Cyrillic, but what to do with Arabic or Chinese characters, for example.

> the question is how to solve this in the PLI image itself? 
 
I think that this is decoded in a certain way at the stage of channel scanning and converted (possibly additionally cached). I can't say exactly. It need to look at the source. In addition, Enigma2 uses a single platform and most channel names use the Latin alphabet (which is why it is not relevant for most). BUT we still have IPTV! And anything can be found here.:) 
Also, when you run the program on a PC, you do not know what OS you are using and what encodings and language settings you'll have.

P.S. As I already wrote, using an id channel avoids many problems and that is why this approach is used in most cases.

chocholousek, 2022-11-20:
> but what to do with Arabic or Chinese characters, for example.
I only support European satellites, sorry.

DYefremov, 2022-11-20:
> I only support European satellites, sorry.

Yes, I know. It was more a rhetorical question in your case. :)
At the moment, displaying piсons by name in most cases working (excluding the nuances with encodings, but I think over time it will be solved. It just needs to modify properly [this function](https://github.com/DYefremov/DemonEditor/blob/c5c482353402738392b5ab8738665af427faabda/app/ui/main_helper.py#L568).) The main, we need to decide, how to assign new picons.

> Additional menu item, additional function or option for conversion (for example, on the "Picons" tab), etc.

I'll be glad to suggestions.

chocholousek, 2022-12-08:
https://picon.cz/dnes-pridany-picony-transparent-snp-220x132-take-do-chocholousek-picons-pluginu/

Hi, SNP picons added... ;)

DYefremov, 2022-12-08:
Hi, Coool! :)
Display and download work!
 
[image]

Just filtering function correction is needed.

As for the names in Cyrillic. I found a working way. But... There are questions to Enigma2 itself.  [The method](https://github.com/openatv/enigma2/blob/d01451e8d75f5d905998b2980eeb0f3bab48055d/lib/python/Components/Renderer/Picon.py#L109) that is currently used to convert names is not suitable. At least it doesn't work for me. Therefore, for now I will leave the conversion in the app unchanged (it is almost identical to the one used in enigma2 itself)

luxorko, 2023-01-20:
looks no issue here and could be close

DYefremov, 2023-01-20:
Fine. If there are suggestions, we can return to this discussion.

closed 2023-01-20 by DYefremov