DYefremov/DemonEditor #130: bouquet path import

luxorko, 2022-10-13:
we can open bouquet list from any folder name, but demon will Import to stb bouquet list always from basic path "demon/datadata/default

DYefremov, 2022-10-13:
If I understand correctly, there is no bug here. So it should be. All data is copied and changed in the profile folder. This is made on purpose so as not to change the data source. To do this, you can use the "Save as..." menu.

luxorko, 2022-10-13:
you can use the "Save as..." menu.

luxorko, 2022-10-13:
of course can use "save as" but bouquet list will be anyway import only from  folder : "demon/datadata/default

its confused if you store your new list : Save as to folder "13.10.2022" and imported list will from location "demon/datadata/default"

DYefremov, 2022-10-13:
> its confused if you store your new list

Hm. I'm also a bit surprised.... :) OK.
In this program, there are 2 options for working with lists.
1. You can download the **FULL** list of settings (bouquets, services) from the receiver or **OPEN** from an external folder.
After clicking "Save" the current data in the profile folder will be overwritten! 
2. You can **IMPORT** data! You have an empty list or you already have data in your list of channels or bouquets. 
Using the import function, you can ADD (combine) data to existing ones from several sources. You can import one bouquet or several services from one folder, then from another folder, then import several services from the Internet, then from another folder etc. I think, you **don't WANT** changes in all sources from which you **IMPORTING** some data, but save the list you **combined in the current profile folder** and then load it into your receiver. I also recommend that you refer to the Important section of the [README](https://github.com/DYefremov/DemonEditor#readme).

> list will be anyway import only from folder : "demon/datadata/default

So far, I've briefly tested the Linux and Windows versions of [this release](https://github.com/DYefremov/DemonEditor/releases/tag/3.0.0-b1). Didn't notice any problems. I will check again if possible.

luxorko, 2022-11-12:
version 3.1.0
path working correctlly, you can change choose any location - Thank you
0345-a6c5-4560-a007-24d94241b64b.JPG)

closed 2022-11-13 by DYefremov