DYefremov/DemonEditor #131: iptv card not showing picon

luxorko, 2022-10-15:
iptv card not showing picons, reference there and match; picon downloaded demon, 

import to stb (picon+iptv bouquet)  picons correctlly showing on TV screen 

[image]

DYefremov, 2022-10-15:
How can I reproduce it?

[image]
[image]

Please, show how looks the reference of one of the channels and name of picon file also.

luxorko, 2022-10-15:
in windows 10 - reference on iptv channel not match (not showing) same as on picon , but when you send bouquet to STB all showing correct! please see screen shot haw reference looks 

[image]

luxorko, 2022-10-15:
forget mentioned references are copy from epg configuration 
also showing just part of reference see picture

[image]

luxorko, 2022-10-15:
can not change reference manually, in iptv or channel bouquet....did you blocked this options

luxorko, 2022-10-15:
did change reference in bouquet , still cannot see picon
only first number in ref. are diferent 4097 vs 1 should not cause problem ?

[image]

DYefremov, 2022-10-16:
> only first number in ref. are diferent 4097 vs 1 should not cause problem ?

Yes, that's the cause. Just drag and drop the desired picon from the picons  list (Picons->Explorer tab) to the selected services in the bouquet list, or vice versa the selected services to the desired picon. Your case revealed an important flaw in work with picons. This can be fixed, but... I think it should be combined with [this](https://github.com/DYefremov/DemonEditor/issues/128).

luxorko, 2022-10-16:
`This can be fixed, but...`
waiting form new version for w10

luxorko, 2022-10-16:
` Just drag and drop the desired picon from the picons list (Picons->Explorer tab) to the selected services in the bouquet lis`
if you drag picon to iptv channel will be created other picon with reference of iptv 4097_1_....  Dont think is right way 
[image]

DYefremov, 2022-10-16:
>  if you drag picon to iptv channel will be created other picon with reference of iptv 4097_1_.... Dont think is right way

It's all right here. The question is only to the first digit, but it is also acceptable.

luxorko, 2022-11-12:
test version 3.1.0

picons showing for IPTV channels list - Thank you 

[image]

closed 2022-11-13 by DYefremov