DYefremov/DemonEditor #132: epg configuration

luxorko, 2022-10-16:
problem with service reference number :

iptv channel have default reference 4097_0_1_0_0_0_0_0_0_0
[image]

then created epg configuration : 
[image]

like you see on right size table - channel showing only part reference  3462:3E9:1:C00000:0:0:0  (this make all prob;em?) without first 3 digit (1:0:1F:) ; 
and after drag to left side table to iptv channel "demon" do automaticly add first 3 digit (4097_0_1) instead (4097:0:1F)

could you change "algoritmus" showing 8 digit (0:1F:3462:3E9:1:C00000:0:0:0) reference on left side on table and to add only first digit "4097" to completed reference to 4097:0:1F:3462:3E9:1:C00000:0:0:0

thank you

DYefremov, 2022-10-16:
It's all ок here.

DYefremov, 2022-10-16:
> channel showing only part reference

It's all ок here.

> could you change "algoritmus"

For me this is not the right way.

luxorko, 2022-10-16:
@Nothing prevents you from setting your own values on import or after.

where is this setting please?

DYefremov, 2022-10-16:
[image]
Or context menu -> IPTV -> 
Or directly in the m3u import dialog.

luxorko, 2022-10-16:
ok but not working in my cane or we not understant each other
so make test plase
epg configuratuion  - and manually move channel HD and then check reference please

in my case 
 if hd channel have reference " 1:0:19:EDC:7D8:2:11A0000:0:0:0" and during epg conf. i move to iptv channel then will showing reference "1:0:1:EDC:7D8:2:11A0000:0:0:0"....third number in ref. "19"-meaning channel is HD
but demon just change third number to "1"

luxorko, 2022-10-16:
definition refernces:

Service reference 1:0:19:379C:14B4:FBFF:820000:0:0:0 ---> A:B:C:D:E:F:G:H:I:L

A : is a default value and it is always equal to 1
B : this is also a default value, and it is always equal to 0
C : service type: TV = 1, Radio = 2, TV HD = 19 , UHD=1F
D : service Id
E: Trasponder Id
F : NId, Network Id
G : Namespace
H : default value, always = 0
I : default value, always = 0
L : default value, always = 0

DYefremov, 2022-10-16:
Since for IPTV you need in this case D, E, F, G only.
I repeat:
> It's all ок here.

luxorko, 2022-10-16:
but is working very wll for changing reference. Than also no problem to see EPG and picon in receiver.

if you will adapt - accpet change "C" then will work for every type of channels 

please change your mind

DYefremov, 2022-10-16:
You obviously didn't read carefully:
> 1. This is not possible for some cases.

> This dialog is for configuring the EPG, not changing the service type.

You can cook pasta in a skillet, but it's better in a saucepan.:)  I have already answered you about picons.

luxorko, 2022-10-16:
might "create - add " other function for reference to IPTV?
in past dont see any software or plugin with this function - parsing satelite reference to iptv channels

`I have already answered you about picons.`
yes, see it

luxorko, 2022-10-16:
workin for each channel and not for whole bouquets of iptv :(

DYefremov, 2022-10-16:
Full matching is possible (for whole list), but not necessary. So in some cases (when changing the type -> first number), your IPTV channels will simply stop working. But it's possible anyway.

luxorko, 2022-10-16:
will be this in new update?

DYefremov, 2022-10-16:
I won't say for sure. A list of tasks that still need to be done has accumulated. 
 Perhaps I will add, as you ask, in the form of an experimental option. Perhaps even in the current dialog. I'll just add one more optional switch. :) I should think. Although a separate dialog, for example, as for the list configuration, may be a more correct option. Anyway, it's not a priority. It is task for experimenting only, when you have free time and desire. :)

luxorko, 2022-10-16:
"test" with epg configuration workin very well (reference parsing work)) with normal channel , like see if will be just add parameter "C" then is new functionaly here .100%
not sure how hard it is

luxorko, 2022-10-16:
will looking on your git hub, when new version realising, will give you feed back.
thank you for your hard work .

luxorko, 2022-10-23:
`Look closely at the context menus! Copy reference (on sat channel)-> FAV list - Context menu -> IPTV - Assign reference
Maybe this will pass. :)`

its faild, reference not assign to iptv - win 10

DYefremov, 2022-10-23:
It works for me as intended. If a service type is needed, then its also not used here.

DYefremov, 2022-11-11:
> its faild, reference not assign to iptv

Should work in the next version (including setting the service type).
Also, as you wanted, I added a service type to the EPG configuration dialog. Now, if the EPG source is the main list of channels, then the type of service will be taken into account when assigning a reference. I did not additional options or settings. Hope this will really helpful for you. :) **Fully not tested!**

luxorko, 2022-11-12:
epg parse showing "parsing" correct ref.number
[image]

ref. number also showing right ref. number in bouquet list

[image]

but channel detail and saving bouquet.tv save wrong ref. number 4097_0_1...instead 4097_0_19
 
[image]

DYefremov, 2022-11-13:
> but channel detail and saving bouquet.tv save wrong ref. number 4097_0_1...instead 4097_0_19

Ok.Thanks, I'll check.

luxorko, 2022-11-19:
great , could you update for new test version please

DYefremov, 2022-11-20:
I think a little later. Not everything is ready yet.

luxorko, 2022-11-25:
confirm is working, thank you for update

closed 2022-11-25 by DYefremov