DYefremov/DemonEditor #133: Extract picon

mapi68, 2022-10-18:
Latest DE for Windows when download picon:

18-10-22 02:42:15 Downloading: https://picon.cz/download/1105
18-10-22 02:42:16 Extracting: 1105
18-10-22 02:42:16 PiconsCzDownloader [extract] error: b'\r\nERROR: C:\\Users\\max\\DemonEditor\\picons\\DM7020HD\\1105.7z : Cannot open the file as archive\r\n\r\n\r\n'
18-10-22 02:42:16 Error: b'\r\nERROR: C:\\Users\\max\\DemonEditor\\picons\\DM7020HD\\1105.7z : Cannot open the file as archive\r\n\r\n\r\n'

In Debian it's ok,

DYefremov, 2022-10-18:
Win 10:

[image]

I'll test later on Windows 11 also.

mapi68, 2022-10-18:
It's strange because before today it worked very good (tested in Win10).

DYefremov, 2022-10-18:
There may be problems with the archive itself. It cannot be ruled out that the  picons author and his service may also have problems. 
Tested on Windows 11:
[image]

DYefremov, 2022-10-19:
Do a little experiment. Just install the [latest 7zip](https://www.7-zip.org/a/7z2201-x64.exe) on your system. Without making changes (replacing) in the program files!

DYefremov, 2022-10-19:
> It's strange because before today it worked very good (tested in Win10).

> Same error....

After some experimentation, I was able to reproduce your case on Windows 10 with **en-us language and locale**!
After installing 7 zip on the system, the error disappeared. But before that, I was playing with regional settings.
[image]

Perhaps it will help.

mapi68, 2022-10-19:
> After installing 7 zip on the system, the error disappeared.

Yes, after installing it works.

[image]

Maybe this is a problem after update to Windows 10 2022H2?

DYefremov, 2022-10-19:
I think you have already answered here: )

> before today it worked very good

mapi68, 2022-11-14:
With Windows 3.1.0 Alpha the same problem of 7zip picons extractor. 
But if I have 7zip installed in WIndows ([full software](https://www.7-zip.org/a/7z2201-x64.exe)) it extract picons. If I uninstall 7zip stop works!

DYefremov, 2022-11-14:
No changes have been made in this version. I think it's just missing a certain library.  It needs to experiment. That is why I didn’t really want to include 7zip in to the Windows build, but always used the external option. :)

DYefremov, 2022-11-14:
Thanks. Updated!

closed 2022-11-14 by mapi68