DYefremov/DemonEditor #134: not sure how to call this

luxorko, 2022-10-22:
if you change bouquet name ==are not keptkept same bouquet name in enigma setting...menining if you created boutt name "my stuff" i dameon , but will be save in "enignma 2" like "bouquet.dbe.129"  why cannor be have same name like name you are created?

DYefremov, 2022-10-23:
[image]

Works for newly created bouquets only!

luxorko, 2022-10-23:
working like you said, i just changed bouquet name and this way not workin, could be in future?

DYefremov, 2022-10-23:
> could be in future?

**No.** This is done specifically to prevent problems with the use of some Enigma2 features.

closed 2022-11-11 by DYefremov