DYefremov/DemonEditor #136: channels mark as "new"

luxorko, 2022-11-15:
some channels are mark as "NEW" and they are colored. - but its very hard to go throught list and "unmark" them.
 Any option to unmark ALL "new" channels in one click, please

[image]

[image]

DYefremov, 2022-11-15:
Yes, I think we can add a separate item to the context menu.

DYefremov, 2022-11-15:
In addition, if it interferes a lot, you can turn off this option or change the color for these channels.

DYefremov, 2022-11-16:
Done! Works for selected services only.

[image]

 **Not fully tested!**

luxorko, 2022-11-25:
yes, working like you said  just " selected services only"
do save channel and open again without "NEW" flag
thank you

DYefremov, 2022-11-25:
Fine. We can close.

closed 2022-11-25 by DYefremov