DYefremov/DemonEditor #138: players not play iptv streams in linux

luxorko, 2022-11-24:
mpv player not work
`Traceback (most recent call last):
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 108, in on_stop
    self.stop()
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 240, in stop
    self._player.stop()
AttributeError: 'MpvPlayer' object has no attribute '_player'
Traceback (most recent call last):
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 108, in on_stop
    self.stop()
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 240, in stop
    self._player.stop()
AttributeError: 'MpvPlayer' object has no attribute '_player'
`

vlc player not work
`Traceback (most recent call last):
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 108, in on_stop
    self.stop()
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 442, in stop
    if self._is_playing:
AttributeError: 'VlcPlayer' object has no attribute '_is_playing'. Did you mean: 'is_playing'?
Traceback (most recent call last):
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 108, in on_stop
    self.stop()
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 442, in stop
    if self._is_playing:
AttributeError: 'VlcPlayer' object has no attribute '_is_playing'. Did you mean: 'is_playing'?
`
gst player not work 
`Traceback (most recent call last):
  File "/home/ja/Downloads/DemonEditor-master/app/ui/playback.py", line 167, in on_realize
    self._player = Player.make(settings.stream_lib, settings.play_streams_mode, self._event_box)
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 170, in make
    return GstPlayer.get_instance(mode, widget)
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 303, in get_instance
    cls.__INSTANCE = GstPlayer(mode, widget)
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 292, in __init__
    self._player.set_window_handle(self.get_window_handle())
  File "/home/ja/Downloads/DemonEditor-master/app/tools/media.py", line 120, in get_window_handle
    return self.get_window().get_xid()
AttributeError: 'GdkWaylandWindow' object has no attribute 'get_xid'. Did you mean: 'get_width'?
`

luxorko, 2022-11-24:
1 - ubundu 22.04.1 amd64
2- installed VLP player and MPV player trought terminal

DYefremov, 2022-11-24:
Then everything should work. I have Linux Mint 21 which is based on Ubuntu 22.04. All three options.
Just install the necessary packages from the **official** repository (**no Snap or Flatpak**). For Ubuntu, you may need to install the _libmpv-dev_ package if you plan to use MPV.

`sudo apt install  libmpv-dev`

DYefremov, 2022-11-30:
I made briefly test. All three variants are working.

[image]

Just configure and install the necessary packages correctly.

closed 2022-12-04 by DYefremov