DYefremov/DemonEditor #140: snr in dB

majortom9, 2022-11-27:
Hi, I really like what you've done here. The player plays nice and smooth
on a 10 yr old laptop. Nicely done.
What I would like to request is for the SNR to be displayed in decibels,
as the percentage isn't very useful for satellite reception.

I have poked around a bit in python and found a way to display this for me. 
Let me know what you think.

`diff --git a/app/ui/control.py b/app/ui/control.py
index 32f2506..3b5eaff 100644
--- a/app/ui/control.py
+++ b/app/ui/control.py
@@ -243,10 +243,11 @@ class ControlTool(Gtk.Box):
 
     def update_signal(self, sig):
         snr = sig.get("e2snr", "0 %").strip() if sig else "0 %"
+        snrdb = sig.get("e2snrdb", "0 dB").strip() if sig else "0 dB"
         acg = sig.get("e2acg", "0 %").strip() if sig else "0 %"
         ber = (sig.get("e2ber", None) or "").strip() if sig else ""
         # Labels.
-        self._snr_value_label.set_text(snr)
+        self._snr_value_label.set_text(snrdb)
         self._agc_value_label.set_text(acg)
         self._ber_value_label.set_text(ber)
         # Level bars.
[image]
`

DYefremov, 2022-11-28:
Hi,
Hm. Maybe both values? For example like this:
[image]

DYefremov, 2022-11-28:
Yes, added this variant. If there are other proposals, then ready to discuss.

majortom9, 2022-11-28:
Perfect...Yeah I see how you implemented that in the development branch. Yes, that'll work for me.
I have really just started using this the other day, anything else jumps out at me I will post here.
Sidenote, using Arch Linux here (Manjaro actually). So far I haven't been able to successfully test it with mpv set as
the player, but setting for VLC or gstreamer both seem to play fine for me. Probably something on my end.
mpv is 

```
DemonEditor]$ pacman -Qs mpv
local/mpv 1:0.35.0-2
    a free, open source, and cross-platform media player
DemonEditor]$ pacman -Qs vlc
local/vlc 3.0.17.4-11
    Multi-platform MPEG, VCD/DVD, and DivX player
DemonEditor]$ pacman -Qs gstreamer
local/gstreamer 1.20.4-3
    Multimedia graph framework - core
local/lib32-gstreamer 1.20.4-1
    Multimedia graph framework (32-bit) - core
local/phonon-qt5-gstreamer 4.10.0-3
    Phonon GStreamer backend for Qt5
```

DYefremov, 2022-11-28:
> So far I haven't been able to successfully test it with mpv

I don't have arch (I'll try to check later in the virtual machine), but I think it should work also. Maybe a package is missing. For example something like 'libmpv-dev'. Also, perhaps it's time to update the mpv module in the program itself. I also recommend checking the work on [this app](https://github.com/DYefremov/TVDemon). There mpv module is updated to last. If it works there, then it is desirable to create a separate issue with this problem.

majortom9, 2022-11-28:
yeah I agree, u can close this issue. I will play around more, and check out that other app.

DYefremov, 2022-11-28:
Ok. Closed.

closed 2022-11-28 by DYefremov