DYefremov/DemonEditor #142: Create and edit profiles

Bender2403, 2022-12-16:
when i try to create or edit profiles under settings it aborts as if i click save at the bottom.

my system is windows 10 64 bit

mapi68, 2022-12-17:
Can you post file config.json?
It is in your home_directory\.config\demon-editor

Bender2403, 2022-12-17:
{
    "version": 2,
    "default_profile": "default",
    "profiles": {
        "default": {
            "setting_type": 0,
            "host": "192.168.100.37",
            "port": "21",
            "timeout": 5,
            "user": "root",
            "password": "Meemann1981",
            "http_port": "80",
            "http_timeout": 5,
            "http_use_ssl": false,
            "telnet_port": "23",
            "telnet_timeout": 5,
            "services_path": "/etc/enigma2/",
            "user_bouquet_path": "/etc/enigma2/",
            "satellites_xml_path": "/etc/tuxbox/",
            "epg_dat_path": "/etc/enigma2/",
            "picons_path": "/usr/share/enigma2/picon/",
            "hosts": [
                "192.168.100.37"
            ],
            "epg_options": {
                "xml_source": false,
                "use_web_source": false,
                "local_path_to_xml": null,
                "url_to_xml": "i.mjh.nz/PlutoTV/de.xml.gz",
                "enable_filtering": false,
                "epg_dat_path": "F:\\DemonEditor\\data\\default\\epg\\",
                "epg_dat_stb_path": "/etc/enigma2/",
                "epg_data_update_on_start": false
            }
        }
    },
    "v5_support": false,
    "http_api_support": true,
    "enable_yt_dl": false,
    "enable_send_to": false,
    "use_colors": true,
    "new_color": "rgb(255,230,204)",
    "extra_color": "rgb(179,230,204)",
    "fav_click_mode": 0,
    "profile_folder_is_default": true,
    "records_path": "C:\\Users\\meema\\DemonEditor\\recordings\\",
    "display_epg": true,
    "backup_before_save": true,
    "backup_before_downloading": true,
    "play_streams_mode": 0,
    "stream_lib": "mpv",
    "main_list_playback": false,
    "language": "de_DE",
    "load_last_config": false,
    "show_bq_hints": true,
    "show_srv_hints": true,
    "default_data_path": "F:\\DemonEditor\\",
    "default_backup_path": "F:\\DemonEditor\\backup\\",
    "default_picon_path": "F:\\DemonEditor\\picons\\",
    "recordings_path": "F:\\DemonEditor\\recordings\\",
    "activate_transcoding": false,
    "active_preset": "720p TV/device",
    "list_picon_size": 32,
    "tooltip_logo_size": 96,
    "list_font": "Normal",
    "dark_mode": true,
    "alternate_layout": false,
    "is_themes_support": false,
    "theme": null,
    "icon_theme": null,
    "enable_experimental": false,
    "use_http": true,
    "remove_unused_bouquets": true,
    "keep_power_mode": true,
    "compress_picons": false,
    "force_bq_names": false,
    "enable_yt_dl_update": false,
    "window_size": [
        874,
        560
    ],
    "data_paned_position": 987,
    "fav_paned_position": 472,
    "backup_tool_window_size": [
        561,
        320
    ],
    "services_update_dialog_window_size": [
        1680,
        1027
    ],
    "last_config": {
        "last_profile": "default",
        "last_bouquet": "Pluto TV - pluto_tv All Channels"
    },
    "epg_tool_window_size": [
        1680,
        1027
    ],
    "picon_paths": [
        "/usr/share/enigma2/picon/",
        "/media/hdd/picon/",
        "/media/usb/picon/",
        "/media/mmc/picon/",
        "/media/cf/picon/"
    ]
}

mapi68, 2022-12-17:
Try to delete file and start DE, so it can create new config.json.

Bender2403, 2022-12-17:
unfortunately it didn't work, do you happen to have log files?

the installer version

can i edit the config file manually with notepad++

mapi68, 2022-12-17:
{
    "version": 2,
    "default_profile": "default",
    "profiles": {
        "default": {
            "setting_type": 0,
            "host": "127.0.0.1",
            "port": "21",
            "timeout": 5,
            "user": "root",
            "password": "",
            "http_port": "80",
            "http_timeout": 5,
            "http_use_ssl": false,
            "telnet_port": "23",
            "telnet_timeout": 5,
            "services_path": "/etc/enigma2/",
            "user_bouquet_path": "/etc/enigma2/",
            "satellites_xml_path": "/etc/tuxbox/",
            "epg_dat_path": "/etc/enigma2/",
            "picons_path": "/usr/share/enigma2/picon/",
            "hosts": [
                "127.0.0.1"
            ]
        }
    },
    "v5_support": false,
    "http_api_support": true,
    "enable_yt_dl": false,
    "enable_send_to": false,
    "use_colors": true,
    "new_color": "rgb(255,230,204)",
    "extra_color": "rgb(179,230,204)",
    "fav_click_mode": 0,
    "profile_folder_is_default": false,
    "display_epg": false,
    "backup_before_save": true,
    "backup_before_downloading": true,
    "play_streams_mode": 0,
    "stream_lib": "mpv",
    "main_list_playback": false,
    "language": "en_US",
    "load_last_config": false,
    "show_bq_hints": true,
    "show_srv_hints": true,
    "activate_transcoding": false,
    "active_preset": "720p TV/device",
    "list_picon_size": 32,
    "tooltip_logo_size": 96,
    "list_font": "Normal",
    "dark_mode": false,
    "alternate_layout": false,
    "is_themes_support": false,
    "theme": null,
    "icon_theme": null,
    "enable_experimental": false,
    "use_http": true,
    "remove_unused_bouquets": true,
    "keep_power_mode": false,
    "compress_picons": false,
    "force_bq_names": false,
    "enable_yt_dl_update": false,
    "window_size": [
        792,
        560
    ],
    "data_paned_position": 0,
    "fav_paned_position": 0
}

Bender2403, 2022-12-17:
unfortunately everything didn't work out but ok it's a beta. but it definitely runs better than dreamboxedit keep it up

closed 2022-12-17 by Bender2403

mapi68, 2022-12-17:
Strange, for me it works good......
Did you use the installer as an administrator or as a user?

Bender2403, 2022-12-17:
as user

DYefremov, 2022-12-17:
Hi,
First of all, you need to try older portable versions!  Based on your description, there is a possibility of a problem with the graphics library. Perhaps [this is our case](https://gitlab.gnome.org/GNOME/gtk/-/issues/5305), but for Windows. In the next build, I'll try to update the set of libraries.

DYefremov, 2022-12-17:
Config is removed due to open password. But it looks OK.