DYefremov/DemonEditor #143: error at start

dacorsa, 2022-12-26:
when i start demoneditor i see this error:

18:51:28 dacorsa@NanoKDE /home/dacorsa $ /app/DemonEditor/start.py 
Gtk-Message: 18:51:31.707: Failed to load module "xapp-gtk3-module"
Traceback (most recent call last):
  File "/app/DemonEditor/start.py", line 28, in <module>
    update_icon()
  File "/app/DemonEditor/start.py", line 9, in update_icon
    with open(icon_name, "r", encoding="utf-8") as f:
FileNotFoundError: [Errno 2] File o directory non esistente: 'demon-editor.desktop'
18:51:31 dacorsa@NanoKDE /home/dacorsa $ 

how solveit?
best regards

DYefremov, 2022-12-26:
Hi,
**What distro** (version) are you using? How do you run the program?

dacorsa, 2022-12-26:
hi,
i use ubuntu 22.04 or 22.10, python10 and start with:
 /app/DemonEditor/start.py
or
python3  /app/DemonEditor/start.py

DYefremov, 2022-12-26:
First of all, you need to read carefully the "Installation and Launch" section of the [README](https://github.com/DYefremov/DemonEditor/blob/master/README.md) .

[image]

For Ubuntu, you may need to install the "_python3-requests_" package.
From terminal:
`sudo apt install python3-requests`

For LTS versions, I recommend using the [PPA](https://launchpad.net/~dmitriy-yefremov/+archive/ubuntu/demon-editor) repository.

Best

dacorsa, 2022-12-26:
Thanks, 
I already installed that package 📦, but I think is a python 10 problem, or of my Python installation

DYefremov, 2022-12-26:
This app uses the default packages (especially on Ubuntu and derivatives). Usually you don't need to install anything. I recommend you read [this](https://github.com/DYefremov/DemonEditor/issues/4#issuecomment-573313785) also.

dacorsa, 2022-12-26:
ok thanks after reinstall : 
pip3 install --upgrade requests
now works

closed 2022-12-27 by DYefremov