DYefremov/DemonEditor #144: description with ":"

luxorko, 2022-12-26:
In the IPTV list - when you change channel name or url (edit)
[image]

after save, bouquet changing -
from 
#DESCRIPTION horror ch+1  
#SERVICE 4097:0:1:....
to
#DESCRIPTION: horror ch+1  
#SERVICE 4097:0:1:....

with colon ":" after DESCRIPTION

DYefremov, 2022-12-26:
Excluding markers, I always have ':'
Something is not working when loading into the receiver or just worries you?

luxorko, 2022-12-26:
other plugin in enigma 2 will showing channels name like - 
DESCRIPTION horror ch+1
removing ":" will showing correct channel name horror ch+1

DYefremov, 2022-12-26:
> other plugin in enigma 2

Are you sure the plugin does this correctly? I have seen both options. Can you tell which one is correct?  No problem to fix it.

luxorko, 2022-12-26:
jedi-epg-extreme

[https://github.com/kiddac/Jedi-EPG-XStream](url)

during browsing iptv list will showing channel name begining "description"

DYefremov, 2022-12-26:
OK. Since this is not principal, I changed it to the desired format (without ':').

luxorko, 2022-12-26:
thank you
could be possible add option "creating add new iptv channel" in the iptv channels list?

DYefremov, 2022-12-27:
So:
[image]
Or like this:
[image]

luxorko, 2023-01-04:
all good, working

closed 2023-01-04 by DYefremov