DYefremov/DemonEditor #145: bouquet editing

risky777, 2022-12-29:
Hi,

I'm trying to use your DemonEditor but there are some problems that might be corrected.

1. In tab "Bouquets" I activate "Filter", and I would like to select only one satellite.
	
	- Problem 1: When I open "Pos" and deactivate "All positions", filter doesn't allow any more selections end closes, and only after that I can open "Pos" and select desired satellite.
	- Correction 1: When I open "Pos" and deactivate "All positions" I would like to be able to select desired satellite in this same step without opening "Pos" once more.

	- Problem 2: When I select desired satellite then name of this satellite in this filter "Pos" becomes invisible.
	- Correction 2: Satellite name should remain visible.
	
2. In tab "Bouquets" in table-context-menu I use "Mark not presented in Bouquets". Could you make this dynamic, so that when I drag channel to bouquets, this channel becomes marked automatically.
	- bouquet editor seems quite buggy so there are other problems here, but I didn't do more tests since this mentioned problem is for me starting point for tests-edits.

Additional information:
I use Xubuntu 22.04, and amiko viper 4k v30. My usuall procedure was to transfer (read/write) data between enigma2 and xubuntu using wine-dreamboxedit, and then do edits in wine-dreamset. Reason for this combination is that there are some channels on 275w-bbc that use unusual audio pid's and dreamset was unable to correctly transfer this information to enigma2 while dreamboxedit was successful in that. DemonEditor seems to handle this transfer correctly, so I don't have to use wine-dreamboxedit anymore. Thanks for that. Now it would be nice if I could also replace wine-dreamset.

DYefremov, 2022-12-29:
Hi,
I'm using Linux mint 21.1 (Mate).
To. p.1 
I have everything working as it should. I can deselect all satellites and then select the satellites I need.  Nothing disappears and there is no need to re-activate 'Pos'. But... If there are no changes in the filtering elements, then after a certain time (2s) the changes are applied and the drop-down list auto-closes. Perhaps this is what you mean. Perhaps there is a sense of increasing the time or other options. E.g.  Don't hide the dropdown list. But when discussing this feature with another user, a wish was expressed to hide this list. Can be discussed. 
To p.2.
Not all users may like it or need it. For example, I don't need it. In addition, this may cause additional performance issues. It may be possible to do this optionally.  This also requires testing and discussion.

DYefremov, 2023-01-01:
> * Satellite name should remain visible

I found what could be the reason. The visibility may depending on used desktop theme.  Try removing [this line](https://github.com/DYefremov/DemonEditor/blob/463702c3710c01a46534164e379a61487a64452b/app/ui/style.css#L40).

risky777, 2023-01-03:
I have complex installation with 2 motorized dishes and 5 lnb's in total so I have a LOT of channels, and I run demoneditor mostly on tablet with intel-z3735 cpu (definitively NOT-FAST). I guess these are constraints :)

1. Satellite-name-visibility: Your proposed change in style.css works, satellite-name remains visible. Thanks.

2. Selection of satellite-positions: When I deselect "All positions" I have to scroll through list of sat-positions (many satellites so usually there is scrolling) and 2 seconds is not-enough time to do another selection. Maybe these 2 seconds can be placed somewhere in settings so it can be modified?
	- I see that when I select satellite and press button "Pos" then dialog is closed and selection is applied, so this button also serves as "ok" button which is very useful in case of longer selection-time (this functionality is enough for me and I just need to disable dialog-timeout).

3. Autorefresh "Mark not presented in Bouquets": I would suggest to put the button for this function next to button "Not in Bouquets". For reasons of performance, make it active only when ONLY one satellite-position is selected, this way demoneditor has to track only channels for one satellite.

4. Bugs in bouquet editor: I've made small tests and there were issues with drag-drop and positioning and saving. But I didn't do proper testings so at the moment I can't write details.

DYefremov, 2023-01-04:
> z3735 
> Xubuntu 22.04
> 4\. Bugs in bouquet editor: I've made small tests and there were issues with drag-drop and positioning and saving.

Hm... Yes, I certainly like to experiment. :) But...
The graphics library used in this program actively uses the capabilities of the graphic system to draw the interface. To render the same drag widget, the [Cairo](https://www.cairographics.org/) library is used. I think not all users want to see the standard drag and drop icon instead of the channel list.Therefore, your "bugs" can be fixed by using an older OS or changing hardware. E.g You can try [this version](https://cdimage.ubuntu.com/xubuntu/releases/bionic/release/) or Debian stable (I had positive feedback about working in this distribution on weak hardware). I have tested with  limited  resources in a VM with Xubuntu 18.04.5 and 22.04.1. Everything works without problems. The only thing I didn't like was the size of some  buttons. Apparently also related to OS-specific styles and themes.

**p.2.** We need to think here. As an option to hide the list when you lose focus on it.

**p.3.** This feature is not related to the filtering function, but is a visual addition only. BUT.. Maybe I will consider some of your suggestions.

risky777, 2023-01-05:
What I meant with drag-drop is not related to visual representation and for me this is not important. Problem was if drop at certain position is correctly accepted and saved there.

Yes, think if something of this suggestions can be implemented. Then I can check drag-drop. And no rush, wine-dreamset works well, it's just that I'm trying to remove wine from usage :)

DYefremov, 2023-01-05:
> Problem was if drop at certain position is correctly accepted and saved there.

As I wrote above, I can't reproduce this problem. On all my OS. But I'll test again.:)
Personally, I don't use drag and drop. This is often slower and not as efficient as using the keyboard and hot keys. For example, moving to the beginning or end of the list, selection with space, etc.

risky777, 2023-01-09:
Don't waste your time on dnd, I will test it more thoroughly if above suggestions are implemented. And if I encounter problems, they will not be related to wayland since xfce still has no wayland :)

closed 2024-04-03 by DYefremov