DYefremov/DemonEditor #146: copy / paste reference

luxorko, 2023-01-04:
epg copy reference - do copy reference
[image]

but when you do paste reference for (IPTV bouquet list) not working 
[image]

DYefremov, 2023-01-04:
This does not work?
[image]

luxorko, 2023-01-04:
unfortunately "assign reference" do nothing

luxorko, 2023-01-04:
only working with channel that have in the front symbol "lock"
[image]

DYefremov, 2023-01-04:
Interesting. Briefly tested on Linux, macOS and Windows 7. Works. I'll try to test later on Windows 10/11. Please specify your OS. Language and Version (Build).

luxorko, 2023-01-04:
win 10, eng

luxorko, 2023-01-04:
correction:find it!!

 you cannot assign reference only from "satellite" what been add to current setting, 
channels reference from adding satellite dont assign reference to iptv channel
ps  - they do copy reference correctlly

DYefremov, 2023-01-05:
Ok. I'll check this.

DYefremov, 2023-01-06:
I have checked with fresh installed (on VM) Windows 10. Everything works as it should. Maybe I'm doing something wrong or I don't understand you correctly. If possible, please describe step by step what and how you do.

DYefremov, 2023-01-07:
After several tests, the only thing I found was the lack of assignment of a service type for an IPTV channel when assigning a reference through the context menu. Corrected. I hope this is what you meant.

luxorko, 2023-01-20:
unfortunatelly not working from all satellite possitions.
- open full setting "vhanibal"  
[-]https://www.linuxsat-support.com/thread/151841-vhannibal-enigma2-setting/?postID=705631#post705631(url)
- add some iptv channel bouquet
- coptv any reference from channel from Astra 28.2 or Astra 19.2
- paste to iptv channel
- nothing imported
if you do same but from Hotbird 13.0 - its working ; ref imported

Any idea?

DYefremov, 2023-01-20:
Yes. I can confirm. This is bug. And very interesting one. Already fixed. I hope.:)

luxorko, 2023-01-20:
perfect, now working

closed 2023-01-20 by DYefremov