DYefremov/DemonEditor #148: epg configuration - more filter

luxorko, 2023-01-20:
could be possible to add more filter for epg configuration - like satellite possition?
[image]
filter by sat. possition you have for bouquets
[image]

please make same for epg configuration card
[image]
Thank you

DYefremov, 2023-01-20:
Ок. I'll think about it. Possible as an multiple filter option (as for the picons tab).
[image]

luxorko, 2023-01-20:
i would say better have filter by satellite possitions, it faster to to find actuall channel for user
etc;
sky are on 28.2uk; 19.2de and 13.0it
if you have iptv list sky it. then choose filter possition 13  
then just auto parse or drag&drop

for - multiple filter option:
if you just wrote sky then will show all sky and need to be very carefull with one is right for user
if you do auto parse then demon will parse first match channel nane.

please make filter by satellite

DYefremov, 2023-01-21:
Ок. For now just a prototype. :)

[image]

luxorko, 2023-01-21:
excellent job, thank you

closed 2023-01-22 by DYefremov

luxorko, 2023-01-28:
hurried, selection by sattelite showing but acctually deamon still make "search and parse" from whole (ALL) channels , instead from selected .
could you look at this please