DYefremov/DemonEditor #150: Time split in EPG

mapi68, 2023-01-23:
It will be useful if time in EPG may be split in 2 columns
[image]

DYefremov, 2023-01-23:
If we are talking about the EPG tab, then personally I do not see any practical sense in this. At least for now. What is actually the benefit of this? In addition, this display option apparently makes sense to put it on the tab itself. Or in the "View" menu.

[image]

But I'm willing to discuss. Perhaps there will be more suggestions or someone else will express their opinion.

mapi68, 2023-01-23:
For example with multiple EPG you can put "End time" in Time order and see whar program is ending now (useful to watch after something from the begin)

DYefremov, 2023-01-23:
Ок. I don't see any big problems in implementing this feature. Perhaps there is a sense to make such view as permanent.

> Perhaps there will be more suggestions or someone else will express their opinion.

DYefremov, 2023-01-24:
Did a few tests. Yes, it will take some work, but... The main question, oddly enough, is how to display time values (in what format).
[image]

With date,  day of the week, etc.

mapi68, 2023-01-24:
[image]

DYefremov, 2023-01-24:
Hmm. Yes, looks ok.

mapi68, 2023-01-24:
Maybe another column with "Length"(in my screenshot: 02h 20m)?

DYefremov, 2023-01-24:
Yes, it's possible. We need to test how will it look.

DYefremov, 2023-01-24:
Probably so:
[image]

DYefremov, 2023-01-25:
Almost done. :) 
It remains to decide on the time format for the duration column. I think it is desirable to use localized names.
Here already existing ones in *.po files. 
[image]

Perhaps it is worth converting to lower case and remove the spaces between the values.

mapi68, 2023-01-25:
But for example in Italian must be:
0 ore
1 ora
2 ore
3 ore
......
It's more simple 00h 50m

DYefremov, 2023-01-25:
> 0 ore
> 1 ora

Yes, for other languages also. As an option, use the first letter in lower case without spaces. For Russian and Belarusian translations  languages this approach is used initially. For me personally, this is not very important, but for other users questions may arise. The day of the week is translated and the hours and minutes are in English.

mapi68, 2023-01-25:
Could you enlarge the "Title" item by a few characters and "Start" and "End" by one character, because the last number remains hidden?
[image]ì

DYefremov, 2023-01-25:
Made. Test again.

mapi68, 2023-01-25:
It's the same, it doesn't change

DYefremov, 2023-01-25:
Strange. It worked for me. Changed through another option.

DYefremov, 2023-01-25:
There is no need for that pull request. My first option **should** have worked! Still, Gtk+ sometimes infuriates me. Looks like I'll have to devote more time [to this](https://github.com/DYefremov/E2Toolkit) project. :)
What distro and version are you using?

mapi68, 2023-01-25:
Debian 11

DYefremov, 2023-01-25:
What desktop?

DYefremov, 2023-01-25:
With this variant the size should adjust automatically.

mapi68, 2023-01-25:
OK, perfect!!!! Yes it's KDE.

DYefremov, 2023-01-27:
Also tested in Debian 11 KDE and Kubuntu.Yes. Looks as working. :) I think we can close.

closed 2023-01-27 by DYefremov